TheZeroBeast/TZB-MMU2S-Firmware #166: Selector blocks manual clearing of failed MMU loads

AbeFM, 2019-11-09:
When I get a bad load, or grind plastic in the MMU, generally I can clear it if I can get the Selector out of the way.

The two buttons are dedicated (I believe?!?) to feeding the filament in and out, and pressing the center moves the selector tantalizingly away but it comes right back before you can clear the error.

Perhaps a long press on the buttons could move selector, or the move to the service position could pause in some circumstances.

Not well thought about but the issue has been nagging me since the generally better handling of things that was introduced a while back.

TheZeroBeast, 2019-11-09:
Hey @AbeFM, in all error states the selector is disengaged to allow manual moving where you please 👍

closed 2019-11-11 by TheZeroBeast

AbeFM, 2019-11-12:
Great. Thanks. I knew that.

I was trying to make a suggestion to make the system easier to use. It is a
Big Pain to move the selector around. Perhaps with a redesigned MMU with a
friendly knob for rotating that shaft, otherwise, it sucks.

It might be more productive to consider ways to use the buttons to increase
the usability of the MMU. Like "go to previous temperature" makes it easier
to use - I have not found rewinding or ejecting filament to be nearly as
useful, which is why I brought it up.

I would ask that you reopen the issue so we can discuss how the two options
effect the user's experience.