TheZeroBeast/TZB-MMU2S-Firmware #167: Initial Load should be Minimized

AbeFM, 2019-11-13:
I'm having issues with blobs on my prime tower. They appear on the INITIAL load - during a filament swap in normal MMU printing, the filament is loaded, then unloaded most of the way as a check for a jammed heat break, then loaded again.

If one decreases (makes more negative) the "extra reload distance" in PrusaSlicer, the filament is not fully loaded, the prime tower is weak, and in extreme cases you can get under-extrusion on your object. However, you still get a large blob on the prime tower which causes crashing as nozzle hits this bump.

[image]

I've been trying to track this down. There's a good chance that I'm doing something else wrong, I've had successful prints in the past.

My thought is that if I decrease the initial load, I'd avoid the blob, and I can use "extra reload distance" to get the material to the nozzle, as the name implies. I feel this is another poor assumption by the PR team - but really I want to fix my issue.

I put forward the suggestion to lower that initial load, but cannot seem to compile it. See next post. But the suggestion will hopefully lead to a discussion about the approach - I'm unconvinced the extreme load is helping anything.

AbeFM, 2019-11-13:
I cloned the latest repository, and selected the RAMBO board, so I would expect this to compile. When I try, however, I get:
`In file included from sketch\Configuration.h:62:0,

                 from sketch\Configuration.cpp:1:

eeprom.h:228:55: error: 'reinterpret_cast<M500_conf*>(20)' is not a constant expression

 static constexpr M500_conf * const EEPROM_M500_base = reinterpret_cast<M500_conf*>(20); //offset for storing settings using M500`

i've gone through and added bits to platform.txt and other voodoo.... I'm not sure why this won't compile. An older version (370 from Prusa) with some tweaks I did compiles fine. 

Here's the old one that compiles:
`#define EEPROM_FIRMWARE_PRUSA_MAGIC 0

#ifdef __cplusplus
#include "ConfigurationStore.h"
static M500_conf * const EEPROM_M500_base = reinterpret_cast<M500_conf*>(20); //offset for storing settings using M500
#endif

#endif // EEPROM_H`

And the new one that doesn't:

`#define EEPROM_FIRMWARE_PRUSA_MAGIC 0

#ifdef __cplusplus
#include "ConfigurationStore.h"
static_assert(EEPROM_FIRMWARE_VERSION_END < 20, "Firmware version EEPROM address conflicts with EEPROM_M500_base");
static constexpr M500_conf * const EEPROM_M500_base = reinterpret_cast<M500_conf*>(20); //offset for storing settings using M500
static_assert(((sizeof(M500_conf) + 20) < EEPROM_LAST_ITEM), "M500_conf address space conflicts with previous items.");
#endif

enum
{
    EEPROM_MMU_CUTTER_ENABLED_enabled = 1,
    EEPROM_MMU_CUTTER_ENABLED_always = 2,
};

#ifdef __cplusplus
void eeprom_init();
bool eeprom_is_sheet_initialized(uint8_t sheet_num);
struct SheetName
{
    char c[sizeof(Sheet::name) + 1];
};
void eeprom_default_sheet_name(uint8_t index, SheetName &sheetName);
int8_t eeprom_next_initialized_sheet(int8_t sheet);
void eeprom_switch_to_next_sheet();
#endif

#endif // EEPROM_H`

Even if I move the preceding line
static_assert(EEPROM_FIRMWARE_VERSION_END < 20, "Firmware version EEPROM address conflicts with EEPROM_M500_base");
to after the statement that throws an error, I still can't get it to compile.

I'm also not convinced 
`#define LOAD_FILAMENT_1 "G1 E80 F400"`
is actually used anywhere, so changing the F400 to F300 may not actually help?

Anyway, I'm stuck, any help is appreciated!

TheZeroBeast, 2019-11-14:
Are you using a Mac with Arduino by chance?

I've been using PlatformIO to compile as Arduino has been playing up on the current OS X release to compile stock or my variant of the firmware.

AbeFM, 2019-11-15:
Windows 10, Ardunio IDE.

Is anyone else using that? That section is different from the earlier one that does compile, but not that line.

Oh, actually, no!
static M500_conf 
becomes
static constexpr M500_conf 

Seems like I could pull that out and see if it compiles.

AbeFM, 2019-11-15:
```
static_assert(EEPROM_FIRMWARE_VERSION_END < 20, "Firmware version EEPROM address conflicts with EEPROM_M500_base");
//abe was: static constexpr M500_conf * const EEPROM_M500_base = reinterpret_cast<M500_conf*>(20); //offset for storing settings using M500
```

That gives me a bunch of warnings about types....

```
sketch\mmu.cpp:113:13: warning: 'mmu_idl_sens' was declared 'extern' and later 'static' [-fpermissive]

 static bool mmu_idl_sens = false;

             ^~~~~~~~~~~~

In file included from sketch\uart2.h:14:0,

                 from sketch\mmu.cpp:6:

sketch\mmu.h:8:13: note: previous declaration of 'mmu_idl_sens'

 extern bool mmu_idl_sens;

             ^~~~~~~~~~~~

sketch\mmu.cpp: In function 'void mmu_continue_loading()':

sketch\mmu.cpp:1112:13: warning: 'void mmu_continue_loading()' was declared 'extern' and later 'static' [-fpermissive]

 static void mmu_continue_loading(void)

             ^~~~~~~~~~~~~~~~~~~~

In file included from sketch\uart2.h:14:0,

                 from sketch\mmu.cpp:6:

sketch\mmu.h:92:13: note: previous declaration of 'void mmu_continue_loading()'

 extern void mmu_continue_loading(void);

             ^~~~~~~~~~~~~~~~~~~~

sketch\ultralcd.cpp:114:13: warning: 'void lcd_language_menu()' used but never defined

 static void lcd_language_menu();

             ^~~~~~~~~~~~~~~~~

Sketch uses 220122 bytes (85%) of program storage space. Maximum is 258048 bytes.
Global variables use 6517 bytes of dynamic memory.

```

So, it compiles.... but.... I dunno, if I should even upload it.

AbeFM, 2019-11-15:
I'm kinda at an impasse. I'd like to try to tweak the loading, but I'm pretty unsure how to make this compile, and if it's safe to use, etc.

Any help in getting the printer to feed less on the initial pre-load would be appreciated. It's causing print problems and I don't see any reason why it needs to do what it does. With compressible filament, I don't get as big a blob, presumably because it doesn't have time to eject.

I'll try another test print with TPU, but I would REALLY appreciate a little help in lowering the one offending number - if someone could make the change and send me a compiled hex or tell me how to do it myself I'd really like to explore this space.

AbeFM, 2019-11-15:
I am doing a test in TPU, and I'm not seeing it eject filament on initial load. I suspect the firmware treating TPU different is part of the difference.

I guess I should try a mixed material print, 2 PLA's, 2 TPU's, and see if TPU-TPU or PLA-PLA is different, or TPU-PLA or PLA-TPU makes a difference.

TheZeroBeast, 2019-11-16:
So still no go with Arduino compiling hey.
When I get some time I’ll review the errors in arduino.

Have you double checked your unloads are working as intended? Maybe do a batch of cold pulls to confirm it’s all clear and try again.

AbeFM, 2019-11-17:
Everything works great as best I can tell. I can bang out spot on TPU stuff all the time, and nothing outside of these blobs on initial load looks wrong.

AbeFM, 2019-11-18:
Ok, I downloaded the latest release versions from Prusa official. I get the same compilation error.

Am I the only person who has these errors? Can anyone else try to compile firmware for the printer using Windows10 and Arduino IDE?

AbeFM, 2019-11-18:
I loaded PR release firmware (3.8.1 I believe) and I DO NOT get the bad loading.

Has anyone else printed an MMU PLA print with the current TZB FW?

[image]
TZB in front, PR in the back.

Yes, even with my original heat-break. Go figure that something which changed when the only thing I touched was firmware was related to firmware.

AbeFM, 2019-11-18:
Looking forward to whatever you find.

Obviously the nozzle will ALWAYS have some material left behind - if not,
there would be no need for a purge tower, it would be a "prime" tower. :-)

The amount that comes out is relatively small, probably ~1mm of filament.

TheZeroBeast, 2019-11-19:
Try this, I found there's a section that loads 3mm more than stock with hard plastics. It's removed in this build.

[TZB-MK3S_EINSy10a-3.0.3-BN2479.hex.zip](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/files/3861232/TZB-MK3S_EINSy10a-3.0.3-BN2479.hex.zip)

AbeFM, 2019-11-19:
Running it with TZB-MMU2S-3.0.2-BN362 & TZB-MK3S_EINSy10a-3.0.3-BN2479, I am still getting blobs. They may be smaller than before, hard to tell yet/at all. :-) Going to try lower loading distance, currently at 85 with -25 extra loading distance. What do you use?

TheZeroBeast, 2019-11-19:
Weird, this is the stock config I'm using.
[image]

AbeFM, 2019-11-19:
Can someone/anyone print either of these? 

[1 USA rings.zip](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/files/3865338/1.USA.rings.zip)
In Red/White/Blue they can be American/French/Russian themed.... I like in PLA but printing in TPU makes it easier to solve.

[3Petals.zip](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/files/3865339/3Petals.zip)
Fast minimal print, I suggest stopping at Z= ~1 mm

AbeFM, 2019-11-19:
[image]

This would be ideal world.... Although in reality you likely wouldn't need to change one of them too often. Similarly, the initial load could be made smaller in firmware.

AbeFM, 2019-11-19:
FYI I have the latest version compiling* now (Thanks for whatever magic you worked!) 

Now I really just want to know what controls that initial load (BEFORE the load-check retract step!!!) distance. ZB mentioned pulling back 3mm from the number, my blobs might be smaller but are still there.
I would like to see that number pulled back another ~5. Can anyone tell me where that change was made and I can play with it myself?

------------------------
*I still get one warning,

```
sketch\ultralcd.cpp:114:13: warning: 'void lcd_language_menu()' used but never defined

 static void lcd_language_menu();

             ^~~~~~~~~~~~~~~~~

```
I suspect this could be fixed with some kinda "ifdef"?

AbeFM, 2019-11-19:
Made the following changes to ultralcd.cpp at line ~4911
```
//abe
#ifdef lcd_language_menu
void lcd_language()
{
	lcd_update_enable(true);
	lcd_clear();
	menu_goto(lcd_language_menu, 0, true, true);
	lcd_timeoutToStatus.stop(); //infinite timeout
	lcd_draw_update = 2;
	while ((menu_menu != lcd_status_screen) && (!lang_is_selected()))
	{
		_delay(50);
		lcd_update(0);
		manage_heater();
		manage_inactivity(true);
	}
	if (lang_is_selected())
		lcd_return_to_status();
	else
		lang_select(LANG_ID_PRI);
}
#endif //abe //lcd_language_menu

```

I just added #ifdef lcd_language_menu and #endif //abe //lcd_language_menu, and it keeps that warning from triggering. I *assume* it won't break anything for those using the LCD, but I don't know.

AbeFM, 2019-11-19:
[image]

I defined a constant, UNDERLOAD, which I subtract from both the first load and the second in "can_load" in mmu.ccp as Rob suggested. I put a #define in Configuration.h

Also I added #ifdef's around the LCD_language_menu to avoid the compilation warning. I changed build number but probably that needs fixing.

I'll upload/commit this if I manage to bend Tortise to my will.

AbeFM, 2019-11-23:
I found the function Rob suggested and added an offset to the initial load and unload. Right now it's only for the Prusa configurations, but I don't see why everyone wouldn't want to have it, with it set to 0 perhaps by default. 

Anyway, I load (and unload) 5 less and I don't get any hint of that blobbing. I could probably run it closer but I don't think loading distance there is worth maximizing.

I've uploaded my source, not sure how to formally present it. At some point others can check it out. :-)

TheZeroBeast, 2019-11-27:
Hi @AbeFM,

The entire point of this initial load is to detect hotend Jams by attempting to insert into cold zone then testing if it lost steps.
So the distance is important and should be at a point where we don't get blobs but also still get into the cold zone for the test.

Was there a smaller amount of change than 5mm that still provided desired results?

AbeFM, 2019-11-27:
I guess I haven't tried. It's a 7% adjustment and adds a lot of reliability - I've been banging out prints and the only issues are missed loads within the MMU, and the lingering complaint about winding the selector out of the way. :-)

You can make the number as small as you want, heck, set it negative and ensure EVERY print fails.... My feeling is the jam detection causes problems and that setting the number for 5 gives a solid margin so it'll never be a problem for anyone else.

Perhaps a better way of putting it - have you had jams that weren't detected because of the 5 mm? I'll bet you you could go for a year without issues.

AbeFM, 2019-11-27:
82 mm gears-to-tip, 70 is the normal number, I'm pushing that to 65, or a net underload of 17. The block and nozzle are 16, so I'm still under the "heat break" (skinny part in the neck). Everything there SHOULD be molten.

TheZeroBeast, 2020-06-02:
Minimised initial load was implemented in release 3.0.3.

closed 2020-06-02 by TheZeroBeast

AbeFM, 2020-06-02:
> Minimised initial load was implemented in release 3.0.3.

I didn't catch that in the release notes, and I didn't install the release. Thanks for doing that!