TheZeroBeast/TZB-MMU2S-Firmware #168: Firmware not working

gizzburn, 2019-11-14:
Hi Robert got round to trying your TZB-MK3S-3.0.2-BN2472-Bondtech-16.hex and the file you sent me for the MMU2s Alpha file remember it would not load the other one, when I upload these two files my printer has a mind of its own, 
When it starts to print after doing the bed level check, it moves the extruder over to left front corner then loads a random filament and purges the filament, it moves back to home, it then starts the bed level process again, then unloads that filament and loads another random filament and starts to print.
It does not matter what I try it always does the sequence.
If I load the Bondtech firmware from there site and the Prusa version for the MMUs2 the printer works as it should selecting all the correct filaments.
Its like its for a different printer

Just for Info I am using the Bondtech BMG extruder so the above is the right one is not?.
Any ideas?...

TheZeroBeast, 2019-11-16:
I’ll review when I get time and get back to you mate 👍

gizzburn, 2019-11-18:
No worries Robert, I just realise how clunky the Prusa FW is to using your version, even the z motors seem louder. I am printing ok on the Bondtech/Prusa FW, but as soon as I load yours the printer does as above.

AbeFM, 2019-11-18:
Is there some chance your filament sensor isn't working right? Try turning off "spool join", etc, see if it cleans up?

gizzburn, 2019-11-18:
No its working fine, I can print with the firmware without touching anything after trying your FW. Its weird because at the very start of the print it does not even do the purge line, it just moves over to the left unloads the filament, then reloads another and then does the bed leveling again. its like the MMU2s is not connected or recognised.? I am sure it thinks its just the Mk3s.

AbeFM, 2019-11-19:
Look at the bottom menu item (About, or System or whatever), printer info - it'll show firmware version, and if you scroll down it'll show if it sees the MMU

Assuming you know, but MMU on far right disables it for the printer. Unlikely either is your issue but it's what I know. :-)

AbeFM, 2019-11-20:
Just an FYI - I turned "filament sensor" on on this most recent build, and before MBL it ejects and reloads filament - it spat it out of the MMU - which of course I missed and had to spend a full minute winding the threaded shaft 1/6th of a turn at a time until I could move the selector over

**PLEASE REOPEN https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/issues/166**

Anyway, it was fine till I turned on "filament sensor". It would be nice if it didn't cause that, but turning it off should get you up and running.

gizzburn, 2019-11-22:
I am just finishing off some long prints, then I will reload your FW and try and see if we can bottom this issue, at the moment its printing well with the Bondtech and Prusa mix FW, but its so clumsy and clonky on operation.. I have disconnected the MMU2s until I finish these long prints. will let you know when I try your FW again.

gizzburn, 2019-12-05:
Hi Robert seeing some issues with the sensor being disabled, its not loading and unloading the same all the time, sometimes it does not fully unload and breaks the filament off, this is random and it could be any filament, apart from this it seems to be working well.

AbeFM, 2019-12-05:
I had a failed unload I couldn't figure out - I believe it was an unload... I found the machine stopped, with the selector on a higher numbered filament, but the lower numbered one still in the tube but not into the extruder.

I don't have a good reason to suspect anything in particular happened - maybe the upper filament sensor triggered after just a bit of unloading. I could think the MMU pulled which the extruder didn't, but the filament wasn't caught on anything when I got to it, so I believe that's out. If it was a failed load, I don't know why it would have been in another position, and I would have expected to see filament shot out the side of the MMU, which I didn't.

Since I had no conclusions, I didn't share, and it's like a 1% kinda event... Anyway, now you've heard about it. If it keeps coming up I'll mention it.

gizzburn, 2019-12-06:
I loaded on the Bondtech firmware last night and did a 7 hour print with exactly the same settings, the print completed fully without issue, there are some bugs in the firmware somewhere or the filament sensor issue needs sorting.

TheZeroBeast, 2020-03-09:
Hi @gizzburn,
during upstream merge to 3.9.0-RC1 I found an issue with M600 triggering before filament loads when running stock start gcodes.
This seams like what you may have been experiencing and should now be resolved in the latest release from earlier today.
Please review and advise.

gizzburn, 2020-03-16:
Hi Robert all good here, just uploaded the latest and a test cube printed with no issues, thanks for your hard work..

TheZeroBeast, 2020-04-17:
Awesome to hear 👍

closed 2020-04-17 by TheZeroBeast