TheZeroBeast/TZB-MMU2S-Firmware #169: Use lower current idler home when in Stealth Mode

AbeFM, 2019-11-23:
**Is your feature request related to a problem? Please describe.**
The homing is now louder than in previous builds, it is too load when everything else is quiet.

**Describe the solution you'd like**
Use quieter homing routine for idler when in Stealth Mode

**Describe alternatives you've considered**
None? Running old firmware?

**Additional context**
I don't think this needs a whole lot more context. :-)

TheZeroBeast, 2019-11-27:
Sorry, I have put many hours into just tuning homing and it needs to be like this to ensure an accurate homing.

AbeFM, 2019-11-27:
Fair enough, its a minor concern.

I can't say I was having issues with poor homing except when my screws were loose. The problem I have is filament missing the selector.

TheZeroBeast, 2019-11-27:
Missing the selector? as in your selector isn't aligning correctly? What selector mod are you using?

closed 2019-11-27 by TheZeroBeast

AbeFM, 2019-11-29:
It's missing... 1% of the time? Maybe 0.5%.

Yeah, the tips just started missing occasionally a few months ago, not sure
why. It seems like if the gap for the razor wasn't there we'd be  a lot
better off.

I'm use JTLX's selector, and been using it for over a year.

I've been a bit suspicious that flexing of the MMU body leads to poor
alignment, but more likely the filament is getting a small curve on the end
and missing the hole because of the gap.

Either way, being able to move the selector back and forth would be a huge
improvement, better than being able to feed filaments towards extruder, IMO.