TheZeroBeast/TZB-MMU2S-Firmware #170: After failed MMU pre-load, retract filament before homing selector

AbeFM, 2019-11-23:
**Is your feature request related to a problem? Please describe.**
When loading filament, if the filament doesn't make it all the way to the FINDA sensor, the MMU re-homes. If the filament is partially in the selector, this can cause the filament to be pulled to the loading position and cause complex jams.

**Describe the solution you'd like**
Once the decision has been made to re-home, retract ~2 cm of filament before.

TheZeroBeast, 2019-11-27:
I'll have a look at solutions.

Although if the FINDA isn't tripped you've ground your filament most likely, but it might prevent the jam at back of SEL.

AbeFM, 2019-11-29:
Generally the situation is:
I'm trying to get the MMU gears to pick up filament, and it takes a while. It finally grabs, feeds a couple CM then gives up.

If it backed up a couple CM before re-homing you would avoid the issue. In short, of the ~5.5 cm from gears to sensor, the last 2.5 cm are all in-selector-but-unsensed, and are bad if you then move selector, at a minimum when you return you'll have bent the filament away from the hole ensuring a failed load, which means you have to manually wind the selector AGAIN, rehome AGAIN, cut off the bent filament tip AGAIN.... :-)

Sucking back 2.5-3 cm before homing shouldn't hurt anything, and at that point you've already given up on feeding it so why not?

TheZeroBeast, 2020-06-02:
I vote for not implementing any changes. If this issue raises it's ugly head you'd need to do maintenance on mmu & knurled gears. Closing off.

AbeFM, 2020-06-02:
That's not what happens. I don't get chewed filament. And there's nothing to do WITHOUT opening it up (in a sealed system) to pull the filament back. At a minimum, a way to pull it back with the buttons needs to be implemented.

If you don't, you break off filament in the selector, or jam the selector on the folded over filament.

You can tell me you *think* it's ground my filament, but in the tens of times it has happened ground filament is basically never the cause. I don't see the downside to retracting before rehoming - the WORST that can happen is you eject it beyond the bondtech - but it started there or it would have fed.

TheZeroBeast, 2020-06-02:
Ok, so how does it not get the filament to the FINDA when loading? I haven't even seen this when printing with FLEX little own with hard plastics that always load true.

AbeFM, 2020-06-02:
Its when I'm first loading the machine on a manual filament change. Trying
to get filament into the mmu before a print. Press the button and the
machine feeds some distance, say 5 inches. If you took a coupe seconds to
get the filament to grip the Bondtechs, then it feeds partway towards the
finda but times out.

At this point there's filament partially extended into the selector but not
all the way to the sensor. When the machine homes it causes effects as
described previously.

Does that make sense?

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

TheZeroBeast, 2020-06-05:
Hmm, I think I understand. I'll try adding a fix 🤔

TheZeroBeast, 2020-06-08:
hi @AbeFM ,

I added an auto retract at each load fail. released with 3.0.4
Its worth noting that the right button in error state retracts and the left extrudes.
centre than re-homes and retries.

closed 2020-06-08 by TheZeroBeast

AbeFM, 2020-06-08:
Awesome! I'll check that out as soon as I get the new firmware. It should
solve the problem.