TheZeroBeast/TZB-MMU2S-Firmware #171: Integrate G81 Center Relative for Nyloc-Mod as Standard-Behavior

XRyu, 2019-11-25:
What do you think about setting the G81-Center-Relative Behavior as a standard?
This is mostly used by Nyloc-Mod-Users which improves Bed Leveling a ton

https://github.com/PrusaOwners/prusaowners/wiki/Bed_Leveling_without_Wave_Springs

TheZeroBeast, 2019-11-27:
Hi @XRyu, has this been requested to be pulled into PRUSA stock FW?

XRyu, 2019-11-27:
No, i dont think so, as its more a thing of the modders and tinkerers to do the nyloc-mod i think :)

XRyu, 2019-12-16:
Close the issue, someone started making an awesome Octoprint Addon for the process, based on the Stock-G81, so any modifications for this would make this harder in the future :)
https://github.com/scottrini/OctoPrint-PrusaLevelingGuide

closed 2019-12-16 by XRyu