TheZeroBeast/TZB-MMU2S-Firmware #172: MMU2S does not work correctly

danielegger1, 2019-11-30:
Hello TZB,

I have an issue with your firmware and I have no idea what`s the root cause, tested several szenarious already.
Everytime when I start a print either in MMU or Single Mode after MeshBedLevelling the printer thinks it must load a filament and runs another MBL afterwards again and asks for the filament I want to pick, I have a video attached. I reflashed the firmware already without any luck. Also the MMU unit is shown as Arduino Leonardo board, is that normal? (I can only flash during reset button action). Using stock slicer parameters.

Any ideas what`s going on?

https://www.dropbox.com/s/modocqm4itlgcq6/Video%2030.11.19%2C%2012%2004%2012.mov?dl=0

danielegger1, 2019-11-30:
Unfortunetly I just found out that MMU printing threw Octoprint doesn`t work at all because at tool change no Unload / Load Action happens, it seems that one of my plugins causes an issue together with the firmware, tested it twice (no video recorded yet) and the same GCode works on the SD Card, will report more details as soon as I find something more out

danielegger1, 2019-11-30:
Forgot to upload image of the flashing menu

[image]

gizzburn, 2019-11-30:
I have the same problem but Robert suggested to turn off the filament sensor, I did this and now the its working as should, but not sure about the effects of having the sensor turned off.
I made a post about a month ago about but have only just got around to re-installing the TZb firmware.

danielegger1, 2019-12-02:
The issue on octoprint seems to come from the printer, Octoprint sends the T0 command but receives a "Duplicate T-code ignored." Message, why does that happen?
[Octoprint.zip](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/files/3912102/Octoprint.zip)
[image]

danielegger1, 2019-12-02:
Thanks gizzburn for your reply, at the moment I will start the prints with the filament in the nozzle until TZB will give a respsonse about the importancy of the filament sensor activated

gizzburn, 2019-12-02:
Just an update I have printed a test piece of about 5 hours and apart from some filament issues's (crappy filament) and some load / unload issues the TZB FW is working well with the sensor disabled, just need Robert to try and sort it when he has the time.

danielegger1, 2019-12-06:
Also an update from my side, I am printing basically 24/7 since the original post and I have to say, god damn this firmware is awesome!
Since the last roughly 2000 filament changes there was just 3 load / unload issues and 1 of that was caused by me :D By the way I disabled the sensor as you mentioned and it didn`t cause any issues yet
Only problem I really have at the moment is that I cannot print threw octoprint

AbeFM, 2019-12-09:
That mirrors my experience, since the adjustment to that initial feed. I
went back to TPU and needed a few degrees of turn on door sensor screw, and
have had a similar number of errors.

TheZeroBeast, 2019-12-25:
Octoprint is fixable by following their guide to setup multitool printers. It’s only come able on the last couple months due to an update they did. I’ve printed a few hundred hours through octoprint with this FW in the last few months without issue.

danielegger1, 2019-12-25:
Thanks for the answer
Can you explain more in details how todo that?

danielegger1, 2019-12-25:
I got it working by changing the following settings to what you can see on the image:
[image]

closed 2019-12-25 by danielegger1