TheZeroBeast/TZB-MMU2S-Firmware #173: mmu security after upgrade to last fw

FredMk3, 2019-12-23:
hi, i don't can use the last 3.0.2 fw because this one make not use good mmu2 because at load print mmu2 load filament on the extrudeur (calibration in center of bondtech ok) but nema ok extrudeur make load filament et this one make stop working mmu2 because of grinding filament.

the only solution is go back to 3.0.0 or 3.0.1 for get working. 

i use this files
TZB-MK3S-3.0.2-BN2472
and TZB-MMU2S-3.0.2-BN362

TheZeroBeast, 2019-12-25:
Can you please go into further detail on this issue you’re having, I don’t follow.

FredMk3, 2019-12-25:
Sorry for my bad English.

On the last rec I cannot print.
I make flash mk3 and mmu, I calibrate Bowden (same as previous version) but when I test to print a file, the filament go to extruder and the motor run for approximately 7 seconds and this make stop print and I do manually push button on mmu for retract the filament. The ir work grate. 
I have test to flash 3 time and same issue.
Return on 3.0.1 work fine. 
I think something is broken in the image. 

i use this files
TZB-MK3S-3.0.2-BN2472
and TZB-MMU2S-3.0.2-BN362

Thanks

AbeFM, 2019-12-27:
Are you REALLY sure the IR sensor is working well?

https://photos.app.goo.gl/qeCGaoDjEYcVkrec6

FredMk3, 2019-12-27:
Yes i'm sure and see it in the support menu.
I actually make a 17hours print on 3.0.1 with no issue. 
I have test this afternoon to flash 3.0.2 another bit same issue 🤔

TheZeroBeast, 2020-03-09:
Hi @FredMk3,

Can you please see if this issue is still present on the release today and advise?

FredMk3, 2020-03-10:
Yes. I make test this week because I think before I don't have time.

I send you if good or same immediately after.

Thanks

FredMk3, 2020-03-15:
All is perfect sir 😉
Work a charm for now.

FredMk3, 2020-03-15:
All work a charm now.
Thanks for your work. 

I close this issue. 😉

closed 2020-03-15 by FredMk3