TheZeroBeast/TZB-MMU2S-Firmware #174: RPI broken

kajsilander78, 2019-12-25:
TZB-MK3S-3.0.2

Bug description:
MK3S wont boot if I power my raspberry. Get just two lines with squares on display.
Not done much more investigation than...

TheZeroBeast, 2019-12-25:
How is your Pi connected? Has your Pi worked on other versions of this FW?

kajsilander78, 2019-12-25:
Its connected to the einsy with the pins, the ones that prusa originally have recommended. Hmm, if it has worked before on this fw, thats a tougher one.... Have not used the Pi for a while, but know it has worked. (its not connected wrong at least ;)  ), think it has worked on this fw also(80%)...but not sure. Sorry.

I have not had chance to investigate more, just noticed when I tried switching on power yesterday, had to unplug power to Pi to get it boot. Now my hobby-room is occupied by guests...

Oh, merry christmas to you!

kajsilander78, 2019-12-25:
Hmm, not sure I have used it since mmu2s upgrade? I'll try to get my hands on it for a quick peek later... Maybe just have to do some config:
https://community.octoprint.org/t/i-installed-octoprint-1-3-12-or-later-and-now-my-mmu2-doesnt-work-properly-the-whole-printer-resets-how-to-fix-this/12962

kajsilander78, 2019-12-25:
Tried changing to 5 extruder setup, no help... Just get 2 lines filled with boxes on display when I flick on power to mk3s.

kajsilander78, 2019-12-25:
Tried flashing down to 3.0.1 version, same result. If I power on printer first it boots fine, then when I boot up PI, it crashes again... not getting any communication, screenshot from octopi:
Connected to: Serial<id=0x689013d0, open=True>(port='/dev/ttyAMA0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Send: N0 M110 N0*125
Send: N0 M110 N0*125
No answer from the printer within the connection timeout, trying another hello
Send: N0 M110 N0*125
There was a timeout while trying to connect to the printer
Changing monitoring state from "Connecting" to "Offline"
Connection closed, closing down monitor

kajsilander78, 2019-12-26:
Guess we can close this for now. Think my pi is broken, could not get it to connect even on original anymore, tried reflashing it with octopi. Now it wont boot. Hehe, more issues... I'll try with some other card. But it would had been nice to know if there is a issue with Rpi or not? Anyone who can confirm it working?

closed 2019-12-26 by kajsilander78