TheZeroBeast/TZB-MMU2S-Firmware #175: Please publish the correct length for PTFE tubing in the hot end.  I would like to make a PTFE cutting jig for this design, as the original Prusa size of 44.2mm is too long.

habitforming11, 2020-04-04:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

TheZeroBeast, 2020-04-17:
Hi @habitforming11,

This firmware doesn't require my upgraded extruder although I do have it on my list to add the correct, shorter length to my Extruder upgrade that's referenced here.

habitforming11, 2020-04-17:
I understand. I was just trying to reach out to you to request that dimension, as I was unable to find it.

gutschke, 2020-04-22:
I measured 41.2 for my teflon tube. Looks fine, but who knows whether it is. The extruder currently works a lot worse than before. Most of the time, I can't load filament it tells me that my heat break is clogged. This is almost certainly an incorrect error message and something else is going wrong. I just haven't been able to find it yet.

TheZeroBeast, 2020-04-22:
Hi @gutschke, thanks for mesuring it.
Correct you are, that is due to the IR_Sensor not being tuned.
When tuned and showing correct states that message is shown when it can't load into the heatbreak (usually due to a jam)

Some tips that may help.
1. Ensure your pin is free, it shouldn't be hard to push in and out.
2. Adjust idler so that when loaded and extruding you don't get false negatives and same when unloaded and extruding

gutschke, 2020-04-22:
I certainly was struggling a lot more with this extruder than with either of the Prusa designs. Yours seems a great design, so not sure why I was having so much trouble. But I think, I finally got it calibrated correctly, after taking it apart a few more times. The pin definitely had to be trimmed and sanded a little bit, and I kept running into issues with the door binding for no good reason (yes, I know about not tightening that bolt too much). The printed parts all look very clean and neat (better than the ones that Prusa printed). So, no idea what was causing all these issues.

But I think the biggest problem is that I can't monitor the status of the IR sensor while loading the filament. That makes calibration really tricky. I now printed a new version of the hat with a clear PLA window for an LED. I am still waiting for my SMD LEDs though; they are stuck somewhere in the mail and might be another week. So, if I ever need to calibrate again, things should be easier.

Also, it would be really cool if you could publish some pseudo code how filament loading is supposed to work. The behavior is non-intuitive. I observed filament going all the way into the extruder. I observed the "support->sensors->IR" status to change, and then it reversed motors, retracted the filament and told me that there was a clog. What exactly triggered this? Which sensor told the printer that the filament wasn't loaded, when it looked to an outside observer that if it was?

In any case, thanks for the great work. I love the fixed firmware and the improved MMU. And I am sure I'll love the extruder now that I seem to have calibrated it. It certainly handled all the tool changes when printing the clear LED window.

AbeFM, 2020-04-22:
FYI, I've had a couple set ups where the sensor turns on-off-on again,
though it got no attention. I ended up having to print a bigger flag and
cut it down. Once I did that, and shortened the load distance a little,
it's been flawless for months on end.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

gutschke, 2020-04-22:
That's a good clue. I wonder if that was happening for me. If this was an official Prusa-sponsored design, maybe the pin should be an off-the-shelf metal part with precise dimensions (e.g. something along the lines of https://smile.amazon.com/uxcell-Stainless-Support-Fasten-Elements/dp/B07MR4VPXM). For an after-market user-provided upgrade, a printed pin is of course fine. And if this is a one-time adjustment, then it isn't too bad. But I am surprised how a an anticipated couple-of-hours upgrade turned into a full-day struggle for me. Thanks to COVID-19, we all have plenty of time for these things :-)

AbeFM, 2020-04-22:
https://photos.app.goo.gl/KZ6GWcQnyC1Dfunm9
I believe there are differences in the hardware between my printer and
TZB's. In the past I was told I had stuff adjusted wrong but it wasn't
that. The same for the loading being too far. As soon as I adjusted it the
problem was fixed and this information was never used to look for possible
issues.

It's not meant to belittle the project, it represents a lot of effort and
it works great after my two tweaks. But nothing else worked and I spent
months making other tweaks.

I would again reiterate that the initial loading distance could be shorter,
I don't see the advantage to having it set so large it borderline doesn't
work.

I can dig around for my lid that has the led hole in it. A window is
overkill, but mainly I was driven by having a through hole led on handhand.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

AbeFM, 2020-04-22:
I feel the printed pin works. It's really tight to get it. I may have
trimmed it a bit. It's been a while but I suspect the pin is at the limit
of working so you have to back the screw off all the way and/or trim the
pin then do that.

Actually, I may have trimmed the door back just a bit to make room.

As I said, once it started working it's literally worked without issue.
I'll try to see if I can figure out what I did. :-)

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

TheZeroBeast, 2020-04-22:
> FYI, I've had a couple set ups where the sensor turns on-off-on again, though it got no attention. I ended up having to print a bigger flag and cut it down. Once I did that, and shortened the load [...]

Hi @AbeFM,
Flag is no longer an issue because it prints on the bed and as long as black plastics are used IR can't get through.

> https://photos.app.goo.gl/KZ6GWcQnyC1Dfunm9 I believe there are differences in the hardware between my printer and TZB's. In the past I was told I had stuff adjusted wrong but it wasn't that. The [...]

I agree an LED like @AbeFM added and @gutschke is planning to add would greatly assist in tuning.
Which load procedures do you want me to elaborate on?

Also, yes the PIN can print tighter at times, usually when I had played with dif. Slic3r profiles (ie. PrettyPETG from Chris) It seams to make all dimensions of the parts a touch larger.
Once tuned the extruder setup is very stable even for TPU which I print automatically between it, PETG & PLA all the time.

gutschke, 2020-04-22:
> Which load procedures do you want me to elaborate on?

In order to test MMU and extruder, I typically use "load to nozzle". There obviously is a lot going on, when doing this. And a lot that can go wrong, if the hardware isn't calibrated correctly. As I learn more about my printer, I am able to deduce some of the steps. But it would make diagnostics easier, if there was a straight-forward "flow chart" of the different things that need to happen. Something along the lines (and I am sure, this is wrong):

1. Nozzle temperature reached
2. MMU retracts filament until FINDA no longer senses filament; then keeps retracting for another 1cm
3. MMU homes selector
4. MMU moves to desired selector position
5. MMU loads filament until FINDA triggers, but no longer than 5s. Otherwise, error at MMU is indicated on LCD
6. MMU loads filament to 90% of bowden length.
7. Extruder motor starts spinning and MMU continues loading at slow speed until IR sensor trips. If this takes longer than 5s, error at MK3 is indicated on LCD and MMU tries to unload filament.
8. Extruder keeps loading filament for another 3s, monitoring missed steps and IR sensor.
9. _... starting from here, my guesses are getting more vague ..._
10. _If at anytime FINDA or IR sensor show an unexpectedly missing filament, ..._

gutschke, 2020-04-22:
@AbeFM:
> A window is overkill, but mainly I was driven by having a through hole led on handhand.

Yes, either a through-hole LED and a hole, or an SMD LED and a diffuser should work just the same. A fringe benefit of the window is that it makes for a great test case for the multi-material capabilities of the printer. If it printed fine, the extruder is clearly capable of loading/unloading materials repeatedly.

I uploaded my *.gcode file to the prusaprinters.org listing, and I am attaching my 3MF file (can't upload that file type to prusaprinters.org). There are 22 tool changes. The part printed fine, but obviously, I haven't tested it yet; I don't anticipate problems though.

[tzbmk3s-hat-with-led.3mf.gz](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/files/4517458/tzbmk3s-hat-with-led.3mf.gz)

AbeFM, 2020-04-22:
>But it would make diagnostics easier, if there was a straight-forward "flow chart" of the different things that need to happen. 
I agree with this. It's a pain to write it up, but it is probably the single thing that makes trouble shooting someone else's procedure difficult.

> 1. Nozzle temperature reached
> 2. MMU retracts filament until FINDA no longer senses filament; then keeps retracting for another 1cm
Not quite following here. Perhaps you meant to say (step 1.5) that the filament is fed forward first? Before the selector is homed I don't think it moves the filament at all.

> 3. MMU homes selector
> 4. MMU moves to desired selector position
> 5. MMU loads filament until FINDA triggers, but no longer than 5s. Otherwise, error at MMU is indicated on LCD
I think it's a distance, not a time.

> 6. MMU loads filament to 90% of bowden length.
> 7. Extruder motor starts spinning and MMU continues loading at slow speed until IR sensor trips. If this takes longer than 5s, error at MK3 is indicated on LCD and MMU tries to unload filament.
> 8. Extruder keeps loading filament for another 3s, monitoring missed steps and IR sensor.
> 10. _If at anytime FINDA or IR sensor show an unexpectedly missing filament, ..._
This is where the on-off-on messes you up.
It's also where overloading can cause issues.
I editing my firmware to load 10% less on this first load and it fixed the remaining issues after fixing the flag.

> Hi @AbeFM,
>Flag is no longer an issue because it prints on the bed and as long as black plastics are used IR can't get through.
The flag in my demo is black PETg. I believe it to be the old style flag, but when I upgraded to the new on-bed flag I STILL had the same issues.

I think the lever needs a few small tweaks, mentioned before, but mainly:
1) The flat spot where the pin hits should be bigger and more perpendicular to the pin. 
2) The flag shape itself maybe needs a different angle. Not quite sure what, but changing it effects the response. I feel my IR sensor is slightly misaligned (on the PCB) relative to your own.

Also the pin:
1) The 'locking in' bump could be smaller, it's never going to fall out.

Lastly, the idler door:
1) The area where the back of the pin hits, and the screw goes through, should be thinned such that the pin NEEDS the screw to be adjusted right, otherwise if it needs to be looser there's nothing you can do. I just checked and mine's shaved down with a razor.

Basically, all of these issues are putting the parts in the middle of the adjustability range instead of on the end of it.

I can run the >90 shoreA stuff, but when I do ninjaflex it's just barely working.
[tzbmk3s-Abelever.zip](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/files/4519137/tzbmk3s-Abelever.zip)
I can't find my LED cover, without more digging.

gutschke, 2020-04-22:
> The 'locking in' bump could be smaller, it's never going to fall out.

I didn't even realize that the bump at the front of the pin was intentional. It made it harder to move, so I smoothed it out. Agreed, it's not going to fall out. If we don't worry about the magnets falling out (which they do on occasion), we absolutely don't need to worry about the pin.

> I just checked and mine's shaved down with a razor.

Same here. Things didn't start to work, until I trimmed the pin by about 1mm. And yes, I then use the small screw to precisely tune things.

Now that I think about it and that I have measured the pin, maybe a metal dowel pin would in fact be a better option. If I measured correctly, then https://smile.amazon.com/uxcell-Straight-Retaining-Stainless-Elements/dp/B01N942DIY should be the correct size.

@TheZeroBeast  what do you think? Is this worth a try, or am I missing something obvious? Is 3mm the correct diameter? Or would 2mm be a better idea? Is the lack of a flag a problem? This is only 30mm long, which is just a little bit shorter than the plastic pin. I think, that's OK; but you'd know better.

AbeFM, 2020-04-22:
There's no reason not to print it. You can "upgrade" anything you want, but
a printed pin won't hold you back.

The VERY best thing you can do it add the LED. That was the first step on
my path to Working. And, um, I still use it constantly, from across the
room I can tell what's going on since it sticks up and all.

gutschke, 2020-04-26:
I received my SMD LED and installed it. And of course, nothing worked afterwards. I guess, taking the hat off and replacing the IR board shifted things just enough that the old calibration was now wrong. And interestingly enough, I could now see the ON-OFF-ON behavior that @AbeFM had mentioned.

A minor adjustment to the screw that pushes the pin, and everything went back to normal. No more ON-OFF-ON. But I agree with @AbeFM, without the LED or some other fast real-time indication of the IR status, that is very difficult to debug.

I wonder what could be done in the firmware to catch and handle this corner case. Overall, I am very impressed how the TZB upgrades have made my MMU2S much more stable. But I am also surprised how difficult it is to calibrate initially.

AbeFM, 2020-04-27:
Another small subtlety: The Bondtech gears are NOT round. I took them out,
measured them, and drew lines on the fat spots. When I close the door I
always check that the high spots are perpendicular to each other so they
sort of average out.

gutschke, 2020-04-28:
Ah, they don't make it easy now, do they. It had never occurred to me to check the gears. I can see how poor alignment due to varying thickness could every so often make it hard for the initial push of the tip of the filament into the gears. Thanks for this suggestion. I'll make sure to measure my gears the next time I take the extruder apart. For now, thanks to the status LED, I seem to have if calibrated very nicely. So, everybody, thanks for all the help, the great design, and the excellent advice on how to spot and correct calibration issues.

AbeFM, 2020-04-28:
What I was finding was that the door would move back and forth even when
empty, or with fixed filament size in it.

The overall problem is the deflection of the idler, especially with softer
materials, is small compared to the natural variations from things like off
center holes. I believe I used a Dial Indicator to find my high spots -
it's not high overall that matters, but the off-centeredness of the central
shaft relative to the outer surfaces and teeth tips.

The LED + TZB's extruder make this all work.

TheZeroBeast, 2020-04-29:
@gutschke, no, not easy but reliability comes with a little effort.

TheZeroBeast, 2020-06-02:
Correct PTFE extruder length published here
https://www.prusaprinters.org/prints/7690-tzb-mk3smk25smmu2s-extruder-mmu2s-no-tower-v2

closed 2020-06-02 by TheZeroBeast