TheZeroBeast/TZB-MMU2S-Firmware #176: MMU2 not seen by MK3

Naugrimohtar, 2020-04-16:
**TZB-MMU2S Release Version**
TZB-MK3S_EINSy10a-3.0.3-BN2480
TZB-MMU2S-3.0.3-BN364

**Describe the bug**
I flashed the firmware per Wiki instructions. Flashing was successful. Upon first reboot I confirmed in the support screen that the versions listed above were correct. I tried to go into the setting menu by going to position 6. The MMU calibrated its position but seemed to reboot after going to position 6. Both LEDs were flashing. I turned the machine off and then back on again. Now everytime I turn it on, the MK3 doesn't 'find' the MMU and in the support menu MMU2 is listed as N/A. That said the MMU powers up, spins the selector, and then sits in a double LED flashing state. 

I can press a button on the MMU to get it to cycle between positions but the MK3 doesn't recognize the MMU.

I tried rebooting 10 times, and on one of those time the MK3 recognized the MMU. It seems to be a power-up sequencing/timing issue? 

**To Reproduce**
I have tried turning the printer off, waiting 10+ seconds, turning it back on. Same results as listed above.
I have a OctoPrint Server on a Raspi. I disconnected it to ensure the problem was with just base hardware. Same issue.
I connected via Octoprint, that causes the printer to reboot, but helps me to capture serial traffic when it turns on. Log text below

**Expected behavior**
I expect the MK3 to find the MMU, open communications, and work as expected ;)

**Screenshots**
None.

**Serial Log**
Connection closed, closing down monitor
Changing monitoring state from "Connecting" to "Offline"
Connecting to: /dev/ttyACM0
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0x710c77f0, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=20.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Send: N0 M110 N0*125
Recv: echo: TZB3.0.3-2480
Recv: echo: Last Updated: Mar  8 2020 19:23:49 | Author: (none, default config)
Recv: Compiled: Mar  8 2020
Recv: echo: Free Memory: 1329  PlannerBufferBytes: 1792
Recv: echo:Stored settings retrieved
Recv: adc_init
Recv: CrashDetect ENABLED!
Recv: FSensor ENABLED (sensor board revision: 03 or older)
Recv: Sending 0xFF
Recv: echo:SD card ok
Recv: echo:busy: processing
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly

Here is a bootup using Real Term instead:
echo: TZB3.0.3-2480
echo: Last Updated: Mar  8 2020 19:23:49 | Author: (none, default config)
Compiled: Mar  8 2020
echo: Free Memory: 1329  PlannerBufferBytes: 1792
echo:Stored settings retrieved
adc_init
CrashDetect ENABLED!
FSensor ENABLED (sensor board revision: 03 or older)
Sending 0xFF
echo:SD card ok
echo:busy: processing
echo:busy: processing

**Additional context**
Hardware differences from Stock MK3.
My original power-supply failed. I replaced with a Meanwell LRS-350.
MMU modified to have festo connectors on the back. Selector replaced with one with magnet in it. MMU has been working 'fine' but obliviously not fine enough hence wanting to try your custom firmware ;)

Is there a better way to capture log data to understand what is going on?

TheZeroBeast, 2020-04-16:
Hi @Naugrimohtar,

It sounds like your in the disable mmu2s state.
Can you move selector to left most with mmu2s buttons then reboot printer?

observatoer, 2020-04-16:
Wait; there is a disable mmu2s state? How to get into this?

TheZeroBeast, 2020-04-16:
Disable mmu2s feature is accessed by going to 6th position (service position) and then rebooting.

Move off that position to one of the 5 filament channels and reboot to reconnect mmu2s.

This allows you to quickly use your printer without the mmu2s, very handy at times

Naugrimohtar, 2020-04-16:
WOW! That seems to be it! What a great feature! Thanks so much! I am *very* excited to explore! I just completed bowden calibration. Awesomeness!

closed 2020-04-16 by Naugrimohtar