TheZeroBeast/TZB-MMU2S-Firmware #177: XYZ Calibration error

sergiofagundes, 2020-05-11:
I have printed the new TZB extruder and update from MK3 MMU2 to MK3S MMU2S latest firmware from TZB also. Before I was using TZB firmware 2.1.8-TZB77 2386 with MMU2S MM-control-01-2.1.8-RC2 with the original prusa extruder.

Before upgrade the calibration passed without problems.

After upgrade there's an error after the printer says about to put a piece of paper during the calibration, the print goes down to the bed and stops whit this error to check the axes.

I have already calibrated the Bowden length as instructed. Also I have noticed the new TZB extruder stay a litle higher to the original prusa when mounted on the x carriage, giving a more height, so I tried to change the Z_MAX_POS to reflect this change.

Maybe this line below can be the cause of the problem:
static constexpr M500_conf * const EEPROM_M500_base = reinterpret_cast<M500_conf*>(20); //offset for storing settings using M500

I have removed the constexpr because the Arduino 1.8.5 or 1.8.12 gives an error, and removing this constexpr make the compilation works.

sergiofagundes, 2020-05-11:
Sorry, I have mistake labeled as a bug, but it's not a bug, it's a help for calibration, problably is just something wrong with my settings, not a firmware bug.

sergiofagundes, 2020-05-11:
After the error to check the axes, I used the move axis to check where the Z axis is, and it shows +5, but the nozzle is much less than this (almost touching the bed), also the X axis is showing +5, and Y axis shows +1.

If I try to do a Calibration / Auto Home, the Z beggins to raise till it touches to top of printer and the X axis goes to left and begins to go to the bed and stops perfectly where the round sensor is located on the bed, so I think there's no problem with PINDA desalignment. After the Auto home, the axis shows Z: +0.2, X: +12 and Y: +6

gutschke, 2020-05-11:
I ran into calibration errors after upgrading the extruder. Turns out, I had removed the zip ties for the heater and thermistor cable. And when I put them pack on, I didn't align them exactly the same way as before. That was enough for the extruder to hit the XZ frame. It was only off by a millimeter or two. But that was enough so that calibration no longer completed. So, make absolutely sure, you don't have any parts that touch the frame, the bed, or the cable housing.

sergiofagundes, 2020-05-11:
I'm testing all day, I think the problem is with the Z axis, It doesn't goes all the all to the top of printer, it only goes a few mms high and the calibration shows a message to continue and use a paper...

gutschke, 2020-05-12:
What happens if you turn off the printer and spin both lead screws by hand to move the Z axis all the way to the top? Does that work? Is it easy, or does it bind on something? How about doing the same thing when moving all the way to the bottom?

I think the stock PTFE tube that connects the MMU2S with the extruder might be just a little bit too short after removing the chimney. I replaced it with a slightly longer one. For testing purposes, you could simply remove the tube completely while calibrating the axes.

sergiofagundes, 2020-05-12:
I got it to work !!!! I followed the Bear document on the web and got it to pass all calibration.

I'm very happy again! The TZB is the only firmware I can get to work with my Prusa since I'm using 3rd party motors and the original Prusa firmware I can't fine tune like TZB. 

Thanks for the help

closed 2020-06-02 by TheZeroBeast