TheZeroBeast/TZB-MMU2S-Firmware #179: MMU2S Heatbreak Jam/Clog Detected

alackofcolor, 2020-06-08:
I see this error state listed in the Wiki, but it doesn't list any information on what causes it or how it's triggered. I'm seeing it with tips that are perfectly shaped, but maybe slightly large in their diameter. Clipping them off usually fixes it, but what's actually being detected to trigger the warning? Is this a stock Prusa error, or one that's added in this firmware?

fromanza, 2020-06-08:
I too have seen this error and the printer usually loads when trying the second time. I assumed there is a stepper current 'overload' detected, like the skipped belt detection stall guard thats working but thats a guess?

TheZeroBeast, 2020-06-08:
Hey guys,

This error state is a stock PRUSA one.

It tries to fit the tip into top of the heatbreak, if it doesn’t fit the filament is no longer present in bondtech gears which then throws this error.

If you’re PRFE section inside extruder is degrading it won’t help this situation.

alackofcolor, 2020-06-08:
How does it leave the Bondtech gears by virtue of being jammed in at the top of the heartbreak? Does it always push the filament forward and then reverse it, causing it to leave the Bondtech gears if it was never able to advance before reversing?

TheZeroBeast, 2020-06-09:
> How does it leave the Bondtech gears by virtue of being jammed in at the top of the heartbreak? Does it always push the filament forward and then reverse it, causing it to leave the Bondtech gears if it was never able to advance before reversing?

Hi @alackofcolor, it always does for all filaments but flex.
Do you understand why it does it?

If the nozzle is jammed and can not extrude there is usually a small amount of material on the heatbreak. The new load comes down and can't push it so gears slip then retract out of gears to catch this situation.

If you don't then it continues to attempt to print, not extruding. Stuffs the print.

@AbeFM, this firmware already has the distance reduced that it attempts to load.

AnimWar, 2020-09-25:
Hi TZB, I really love your firmware and extruder, and hoping you can help me out here.
I repeatedly get this problem even though there is no clog or jam in my extruder. 
My best guess is that it happens because it retracts so far during testing that only the thin tip remains in the extruder gears and doesn't activate the IR-sensor. I was unable to get it to reliably detect the filament in the last bit of retraction.
Where would I find the values/functions in the firmware for this test?
Thanks

TheZeroBeast, 2020-09-25:
Hi @AnimWar,

If you're using the standard e3d v6 that comes with the MK3S I would recommend finding the hardware problem and fixing it.

If it's one of the third party hotends that I haven't used before there could be merit in adjusting the values for the jam det. retinue.

The method is in the printer firmware called **can_load()**, in file **mmu.ccp** on line **1087**
[https://github.com/TheZeroBeast/TZB-MK3S-Firmware/blob/MK3S-TZB/src/mmu.cpp]Linked Here

There are a bunch of comments to help guide you.

TheZeroBeast, 2021-01-11:
Stale thread, if you're still in need of assistance please replay back to this thread.

closed 2021-01-11 by TheZeroBeast

AbeFM, 2021-01-12:
I remember having issues here, and shortening up the load made mine
reliable. I need to dig into it before trying the new firmware, it didn't
work well without it, for me. I'm assuming making that editable over serial
or the menus would be too involved to be worthwhile to work into the
firmware?