TheZeroBeast/TZB-MMU2S-Firmware #181: TZB firmware stuck on boot screen

sergiofagundes, 2020-06-14:
**TZB-MMU2S Release Version**
3.0.4

**Describe the bug**
After installing new firmwares 3.04, the printer stays at "Original Prusa i3 Prusa Research ..." I've upgrade from 3.03.

Steps to reproduce the behavior:
1. Upgraded first the MMU2S firmware (If I try to upgrade the printer firmware first I can't)
2. Upgraded the printer firmware
3. Screen shows verifying 100% and stuck at it.
4. I tried hiting the reset and printer stays at "Original Prusa i3 Prusa Research ..." I've upgrade from 3.03."
5. To fix I need to go back to version 3.03

TheZeroBeast, 2020-06-16:
Hi @sergiofagundes, what printer version are you flashing and are you flashing via PRUSASlicer?

sergiofagundes, 2020-06-16:
I have flashed version 3.0.4 on the MMU2S first. After I flashed the printer (einsy), also version 3.04. I used the Prusaslicer for this. Before, I was using 3.0.3.

I have compiled with Arduino 1.85. Using Arduino I have to change the line bellow on eeprom.h:
static constexpr M500_conf * const EEPROM_M500_base = reinterpret_cast<M500_conf*>(20); //offset for storing settings using M500

I have to remove the constexpr to compile the printer firmware (I have used this also in version 3.0.3).

sergiofagundes, 2020-06-16:
Got it!

Changed the Arduino board from:
Prusa Research AVR Boards
to
Prusa Research AVR MK3 Rambo Einsy

Now it worked, strange is that older versions of TZB till 3.0.3 works with the newer Prusa Research AVR Boards.

The only thing I see strange is that in Support it shows:
Fil. Sensor v.:
unknown state

But in Sensor Info it shows IR sensor changing value (0 to 1) if I manually press the screw to activate the IR sensor.

gutschke, 2020-06-16:
_Fil. Sensor v.: unknown state_

I believe this is the first release of the firmware, where Prusa added code to detect the type of filament sensor that you have. The choices are: new IR sensor, older laser sensor, different unsupported sensor, or none.

Had you bought the printer with this particular firmware pre-installed, it presumably would have tested and saved the correct values in the EEPROM already. But since you are upgrading to the new feature, you need to make the printer go through its self test procedure.

Go into Settings->HW-Setup->FSensor-Detection and run through the initial detection process. Things should work afterwards. At least they did for me, despite the fact that I installed an indicator LED on my sensor.

Prusa warned in the release notes that modified sensors could potentially mess with automatic sensor detection, in cases where the LED resistor values were too small. I guess, I chose wisely.

sergiofagundes, 2020-06-16:
It worked following your instruction. Now it detect it as a 0.3 revision or older.

Thank you

closed 2020-06-17 by sergiofagundes

TheZeroBeast, 2020-06-20:
Good to hear it's working for you mate 👍
I wasn't aware the Arduino board selection had changed either, I only build with PlatformIO.