TheZeroBeast/TZB-MMU2S-Firmware #182: MMU2 setup menu

michaelbeljaars, 2020-07-07:
I am not sure if it is a bug, or if I have just overlooked the option, but in the readme it mentions this functionality:

- LCD Guided MMU2S SETUP menu for more intuitive BOWDEN calibration

which I am not able to find anywhere in the menu structure of my MK3S + MMU2S (TZB-MK3S_EINSy10a-3.0.4-BN2481.hex and
TZB-MMU2S-3.0.4-BN364.hex firmware respectively). I have a hunch that I need to do this calibration, even though it should use the filament sensor to detect the filament, because I hear the bondtech gears grinding on the filament.

peacemaker85, 2020-07-07:
Bowden length calibration is described in great detail on the wiki https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/wiki/Bowden-Length-Calibration :-)

michaelbeljaars, 2020-07-07:
> Bowden length calibration is described in great detail on the wiki https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/wiki/Bowden-Length-Calibration :-)

Great! Thanks! So but no LCD guided option yet?

Great work btw, my MMU2 is already working better than ever before. I had almost given up, but now I have regained faith that I might actually get it to work!

gutschke, 2020-07-07:
> Great! Thanks! So but no LCD guided option yet?

If you updated the firmware for both Einsy and the MMU2S, then the MMU2S can send messages to the LCD. This will be used during Bowden calibration

closed 2020-08-10 by TheZeroBeast