TheZeroBeast/TZB-MMU2S-Firmware #183: Prusa-Firmware 3.9.1

timemaster5, 2020-09-27:
Thanks for a great job. Just a notification that Prusa has recently released a final of 3.9.1.

Ruhkukah, 2020-10-12:
Hi timemaster5 and all, since TZB hasn't updated his firmware to 3.9.1 yet I decided to merge changes of his firmware and Prusa's FW 3.9.1. This is my first attempt with Github and I haven't figured everything out yet, so I did it in my own repository: https://github.com/Ruhkukah/Prusa-Firmware/tree/TZB-Full-Ruhkukah-3.9.1. Basically, I took 3.9.1 files and transferred all TZB's changes meanwhile adapting to the new flavor of 3.9.1 (for example, they made many changes that overlap with the TZB's code so I made these changes too).

I also included one precompiled binary for MK3S with a stock extruder (it is in the folder "Prusa MK3S Stock - compiled binary").

After the update both MK3S and MMU2S work great, as expected. I am now close to an end of 5-filament print with ~350 color changes and I haven't noticed any issues.

timemaster5, 2020-10-13:
thank you for this, I am gonna flash it today 👍

Ruhkukah, 2020-10-13:
You are welcome. Let me know how it works

timemaster5, 2020-10-15:
It looks goo. I see new options in the menu (Preheat -> PC ). Don’t know yet how to test the LA. MMU works great as before on this fw:) so I can’t see any issue so far.

Ruhkukah, 2020-10-15:
Awesome. You can easily test LA by running a calibration code from Prusa: https://help.prusa3d.com/en/article/linear-advance_2252

timemaster5, 2020-10-16:
Thank you for sending that link. I meant how to check the fixed LA15 instead, sorry for the confusion. The LA15 was there already in the previous version of the firmware. But I think it must work as TBZ changes are not related to this part of code anyway.

Ruhkukah, 2020-10-16:
Yes, you are right - his changes do not modify any LA functions. On that note, I am looking to port his MMU FW mods into the Marlin 2 firmware, which I heard is even better than Prusa's one. But for that I need to replace Einsy with SKR BTT board first

timemaster5, 2020-12-07:
@Ruhkukah have you noticed any TEMP RUNAWAYS after flashing your modified firmware? 
I am having these issues quite a lot now, and starting to think that they are not caused by the printer itself. So far I have changed the whole extruder together with thermistor and heat block, and wires and one week ago disconnected a power panic wire from the PSU. But problems are back after a few prints. It could be a broken electronics board, but I haven't heard of anyone with such issue :(

Ruhkukah, 2020-12-07:
Hmm, no issues for me at all. I googled this: https://help.prusa3d.com/en/article/thermal-runaway-temperature-drops_2131
As far as I remember the code for thermal runaways was not modified in the firmware. I suggest that you flash some stock firmware from Prusa and see if it solves your issue. If it does, I'll need to look into the issue but I highly doubt that a firmware is responsible for throwing out this error

timemaster5, 2020-12-07:
Thank you for you response. If you are not having similar issues, than I also doubt it is a firmware thing. Maybe my EINSY board is really bad. It could also be a Beta of Prusa Slicer as I saw the thermal runaway now twice at very similar moment of the same print. This should be easy to confirm just by using the latest stable one.

I think ignore it for now, I was just curious if someone observed something similar so I could stop replacing the electronics :), but it is probably my printer issue.

And thank you for the link.

Ruhkukah, 2020-12-07:
Look, I highly doubt that Prusa Slicer has anything to do with that. I can't think of a way that gcode commands from PrusaSlicer would affect the algorithm of thermal runaway. Just to check - have you done a PID calibration? 
Also, if you have a suspicion that your Einsy is fried then I suggest switch the hotend thermistor to another input port on the Einsy (see here https://help.prusa3d.com/en/guide/4-electronics-assembly_58016#58216). To make it work you should also change an input port in the firmware. Does it make sense?

timemaster5, 2020-12-08:
I didn't know, there are some spare ports, thanks for that, I will try.

My thinking behind the Prusa Slicer problem was that I am using their Alpha release because of some latest features. So I thought maybe some unsafe gcode could get into the final file and cause delays in PID loop or something like that.

I've done PID calibration multiple times, will try again too.

TheZeroBeast, 2020-12-26:
Hi guys, now rebased upstream to 3.9.2, check it out [here](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/releases).

closed 2020-12-26 by TheZeroBeast