TheZeroBeast/TZB-MMU2S-Firmware #185: Updated FW to 3.9.1

Ruhkukah, 2020-10-12:
Hi TZB, I decided to merge changes from your firmware with Prusa's FW 3.9.1. This is my first attempt with Github and I haven't figured everything out yet, so I did it in my own repository: https://github.com/Ruhkukah/Prusa-Firmware/tree/TZB-Full-Ruhkukah-3.9.1. Basically, I took 3.9.1 files and transferred all your changes meanwhile adapting to the new flavor of 3.9.1 (for example, they made many changes that overlap with your code so I made these changes too).

I also included one precompiled binary for MK3S with a stock extruder (it is in the folder "Prusa MK3S Stock - compiled binary").

After the update both MK3S and MMU2S work great, as expected. I am now close to an end of 5-filament print with ~350 color changes and I haven't noticed any issues. Feel free to use it and let me know if there any issues with how I updated the firmware - I am new to this. 

And thank you for your work - I enjoy printing with MMU2S after I upgraded to your FW!

Ruhkukah, 2020-10-12:
Also, I tried to merge changes to your firmware directly from 3.9.1 but there were too many changes. I ended up modifying 3.9.0 with the code from your FW and then updated changes from 3.9.1 to the modified 3.9.0

XRyu, 2020-10-13:
@Ruhkukah can you enable issues on your repo too? as it looks you're going to be the spiritual successor of TZB for now ;) 
I would love to see your FW built with the BMG + Mosquitto Magnum + SE HT Thermistor :)

Ruhkukah, 2020-10-13:
@XRyu done! see here https://github.com/Ruhkukah/Prusa-Firmware/tree/TZB-Full-Ruhkukah-3.9.1/MK3S_Bondtech-16-EINSy10a-SE_HT-Mosquito_Magnum

Let me know your feedback on how it works

Ruhkukah, 2020-10-13:
These are two types of files that Arduino exports after compilation. Since I use Arduino to flash my Prusa directly, I don't know exactly which of these two files to use, so I uploaded both of them. Please let me know which one of these works.

Regarding PlatforIO - I may explore it later this week. For now I wanted to keep as close to stock compilation recommendations as possible to minimize any glitches with the FW

XRyu, 2020-10-13:
Ah alright. Flashing with Octoprint warned me to flash the with_bootloader file, which might corrupt the boards bootloader.
just a question - when the TZB-Firmware was flashed there was the name of the build displayed after starting e.g. "TZB-MK3S-BMGMM" or the like. Now its not :) And in the Support Menu i see only "E3Dv6full", i sadly have not compared what it was before, but shouldn't there be something about mosquitto magnum and the SE HT or the like? :)

Ruhkukah, 2020-10-13:
@XRyu I am confused. Which file did you end up flashing with - without the bootloader or with the bootloader?
Also, what is the name of the build that is displayed after you start the printer? I can't check mine, since I am in the middle of print

XRyu, 2020-10-14:
@Ruhkukah I flashed the file without the bootloader :)
And i was wrong (Octoprint was just faster than me having a look) - It shows "Bondtech-MM-TZB MK3s" right after the start.

danielalexis, 2020-11-13:
@XRyu Hi! i will be updating my repo to include new versions!

WarrenSchultz, 2020-11-17:
@Ruhkukah  I haven't been set up to build firmware for a while. If I do a pull from your source tree, does it compile MK3S_Bondtech-16-EINSy10a-SE_HT-Thermistor automatically? (If you want to upload the hex for that, I can test if you're not set up to do so, otherwise, I'll set up my environment again.)

Ruhkukah, 2020-11-22:
> @Ruhkukah I haven't been set up to build firmware for a while. If I do a pull from your source tree, does it compile MK3S_Bondtech-16-EINSy10a-SE_HT-Thermistor automatically? (If you want to upload [...]

Hi @WarrenSchultz, not sure that I fully understand your question but here is my take on it: you need to compile a branch found here: https://github.com/Ruhkukah/Prusa-Firmware/tree/TZB-Full-Ruhkukah-3.9.1/Firmware
Since you want to use a **MK3S_Bondtech-16-EINSy10a-SE_HT-Thermistor** configuration, you need to copy an appropriate file from the **variants TZB** folder and rename it to **Configuration_prusa.h**. Follow instruction to compile a firmware from [https://github.com/prusa3d/Prusa-Firmware](url). Let me know if something is not clear

WarrenSchultz, 2020-11-23:
@Ruhkukah 
> Hi @WarrenSchultz, not sure that I fully understand your question but here is my take on it: you need to compile a branch found here: https://github.com/Ruhkukah/Prusa-Firmware/tree/TZB-Full-Ruhkukah-3.9.1/Firmware
Thanks for the tips, ran into an issue with paths on the precompiled binaries causing git to not like pulling down to WSL, so I'm looking for a way around that, but if it's something you can fix on your side, you might want to for anyone else. 
Thanks again

[image]

TheZeroBeast, 2020-11-24:
Hey guys, sorry, just saw this thread. I’ll rebase to 3.9.2 on printer FW this week for us 👍

TheZeroBeast, 2020-12-26:
Hi guys, now rebased upstream to 3.9.2, check it out [here](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/releases).

closed 2020-12-26 by TheZeroBeast