TheZeroBeast/TZB-MMU2S-Firmware #187: 3.9.0 cut feature not enables in TZB?

Naugrimohtar, 2020-11-23:
I am looking to enable the MMU2s cutter using the settings that seem to be released in the 3.9.0 Prusa firmware release but I cannot seem to find them in TZB. Is this implemented? For example see timestamp ~12:30 in this video by Chris' Basement -- https://www.youtube.com/watch?v=LpoCYmSb7wo&ab_channel=ChrisRiley

TheZeroBeast, 2020-11-24:
Not a bug mate, different firmware. Possibility to add in the future.

XRyu, 2020-11-24:
@TheZeroBeast Welcome back on board Captain :D Nice to see you back :)

TheZeroBeast, 2020-11-24:
Never gone, just GitHub notifications don’t seam to reach me reliably😂

XRyu, 2020-11-24:
Haha 😂🤣 i know its OT, but are you gonna merge the latest stuff like Ruhkuka did and update your versions? :D

TheZeroBeast, 2020-11-24:
Yes I will be merging, I just haven’t needed to yet from 3.9.0
Let me go review 3.9.1 changes

TheZeroBeast, 2021-01-19:
FYI @Naugrimohtar, I'm coding cut now for testing and release in TZB 3.2.0 release.

Naugrimohtar, 2021-01-19:
Excellent! Thank you!

TheZeroBeast, 2021-01-22:
[Resolved in Release 3.2.0](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/releases/tag/3.2.0)
Please comment if you have any questions.

closed 2021-01-22 by TheZeroBeast