TheZeroBeast/TZB-MMU2S-Firmware #189: Please add a variant for MK3S Bondtech Mosquito SE HT Thermistor

viper2g1, 2020-12-24:
On the supported printers section of the firmware releases page there is no specific configuration for an MK3S with Bondtech, Mosquito, and a Slice HT Thermistor.  

There is a similar option for the Bondtech, Mosquito Magnum, and the Slice HT Thermistor.  

I do not know if those are interchangeable since there are specific different versions for the following:

MK3S Bondtech Mosquito
MK3S Bondtech Mosquito Magnum

Please add a variant for:

MK3S Bondtech Mosquito & SE HT Thermistor

TheZeroBeast, 2020-12-26:
Hey, I'll see if there's much difference with the magnum config and get it added mate.

TheZeroBeast, 2021-01-12:
I've added this into the next release which is immanent.
Testing is happening this week to check on #192 issue then release.

TheZeroBeast, 2021-01-22:
[Resolved in Release 3.2.0](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/releases/tag/3.2.0)
Please comment if you have any questions.

closed 2021-01-22 by TheZeroBeast