TheZeroBeast/TZB-MMU2S-Firmware #191: Pinda Probe Bug?

gizzburn, 2021-01-01:
Hi Robert not if this is a bug, but when I upgraded to the latest version of FW I had some strange errors, one was when the carriage started to move to start the 7x7 bed level checks is shuddered in a couple of spots in the first 4 check's, like it was binding on something.
But after that it would not pass the 7x7 checks, it kept wanting to Calibrate the Z, It would just stay there until I pressed the button to Calibrate Z.

It would go to the top of the frame and home again, but as soon as the 7x7 check started it would throw up the Calibrate Z error.

I had just completed a long multi colour print before I updated the FW with no issues.

I then went back to the old version I was using when I did the multi colour print and all is fine, no errors.

Can you suggest anything.

I have the Mk3s MMU2s  using the Bondtech BMG  PE FW on my Bondtech BMG extruder.

Gizz.

gizzburn, 2021-01-02:
Hi Robert my mistake I realised what had happened, I had put my USB cable in the control box in front of the main loom to the extruder when I plugged it back in after the update, this was enough to cause the problem.
I found it by doing the same thing today when I tried the FW again.

So all is well and no bug..

Keep up the great work.

TheZeroBeast, 2021-01-03:
No worries mate, happy printing.

closed 2021-01-03 by TheZeroBeast