TheZeroBeast/TZB-MMU2S-Firmware #192: Problem loading Filament on a MMU single mode print

eurohell, 2021-01-02:
Hi,

i had to upgrade my printers from 3.0.4 to 3.1.0 because of upgrading SUPERpinda. Now i got a strange behaviour when starting a single-color-print with a cold extruder. in 3.0.4 it was the way, that when you choose the filament slot, the filament will be loaded until the ir sensor is triggered and the MMU LED lights green. Then it waited for heating up the nozzle and then it was loaded into the nozzle.

With 3.1.0 and no other changes, all of my machines are having trouble. The MMU tries to load to a cold nozzle, but when triggering the ir sensor, the MMU pushes on and on, through the extruder pulley straight into the cold nozzle. Then the MMU slot starts flashing red. And if you wait, nothing more happens.
This can only be resolved by unloading the filament (right MMU button), waiting for heating up, entering the MMU setup mode and leaving this mode.

Do you have a workaround for this?

TheZeroBeast, 2021-01-03:
I'll have a look, it's most likely introduced from the upstream pull (AKA, I missed adding my code for that part in or somethingp)

TheZeroBeast, 2021-01-21:
Fixed as above, indeed a regression, release with 3.2.0 of my TZB code.

closed 2021-01-21 by TheZeroBeast