TheZeroBeast/TZB-MMU2S-Firmware #193: Doesn’t retry on a clog/jam

Naugrimohtar, 2021-01-21:
Occasionally I get a clog/jam error and when I go and check, all looks good so I acknowledge, it retires with no issue. When I watch it happen, it doesn’t seem to try to reload itself, I thought it  retries with the standard build

TheZeroBeast, 2021-01-21:
Hey @Naugrimohtar,
I'm not sure if stock tries again but it does have the cutter enabled which cuts and they tries again.

I have just finished testing my implementation of the cutter on selector which will be an option in the next release.

TheZeroBeast, 2021-01-21:
I just reviewed stock code and I had removed retry 3 times and pause for assistance, I will re-enable on the next release, it might help at times.

TheZeroBeast, 2021-01-21:
I just implemented this in my codebase for testing this weekend before release 3.2.0
[TZB-MK3S-FW Commit](https://github.com/TheZeroBeast/TZB-MK3S-Firmware/commit/6ae43bc7e67ca7f5a0e80502f0c1f14dbb82738a)

TheZeroBeast, 2021-01-22:
[Resolved in Release 3.2.0](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/releases/tag/3.2.0)
Please comment if you have any questions.

closed 2021-01-22 by TheZeroBeast