TheZeroBeast/TZB-MMU2S-Firmware #194: M112 with Octoprint on MMU Single Mode

XRyu, 2021-01-21:
Hey TZB, can you have a look if you lost some part in the firmware regarding the Octoprint causing M112? 
I have these more often lately when printing in Single Mode - 
Its mentioned here on 3.9.1 (last point in 3.9.1): https://github.com/prusa3d/Prusa-Firmware/releases

TheZeroBeast, 2021-01-22:
Mahouts be all fixed in my next release as it’s rebased to stock 3.9.3

TheZeroBeast, 2021-01-22:
How often are you getting the error?

XRyu, 2021-01-23:
I 4 attempts yesterday i hd 2, after having it and resetting the printer it
works. i think it has something todo with the printer waiting for new
commands from octoprint and octoprint waiting for the printer to warm up.
after the reset and reconnection it wont take as log to heat up.

I will try to flash to the new version tomorrow :) so much thanks for your
work! :D

TheZeroBeast, 2021-02-05:
Hey @XRyu, did you get a chance to try this out on the latest release of my code?

Also, this is my list of octoprint long running commands, it may play a roll in this error and stopping it.
`G4, G28, G29, G30, G32, G80, M400, M226, M600, M860, M190, M109, M701, M702`

XRyu, 2021-02-11:
Hey @TheZeroBeast, thank you!
After installing the latest release with my first testprint everything worked :)
Sadly i had an incident, got a jam in the heatbreak of my Mosquito Magnum with Prusa PLA, so i had to disassemble etc. When re-assembling i seemingly took to much force, broke the Bondtech Shaft Gear :'D Had to order replacement, after that i also seemed to have broken my HT-Thermistor somehow (MINTEMP) - again - replacement ordered and installed - first day back working today :) As far from now it looks good, error stopped :)

TheZeroBeast, 2021-02-12:
@XRyu, 

Annoying when a bunch of failures happen at once but glad you're now back up and running.

closed 2021-02-12 by TheZeroBeast