TheZeroBeast/TZB-MMU2S-Firmware #195: Min Temp

gizzburn, 2021-01-21:
Hi Robert, would you be able to put an option in the menu to alter the Min temp setting. My printer is in my garage and its been around freezing here in the North UK.
I am having to use a heat gun on my printer to warm it up so it does error with min temp warning.
With a setting that could be tweaked would be helpful.

thanks MJ.

TheZeroBeast, 2021-01-22:
Hey mate, are you using a MK3S/+? If so I can add a customer gizzburn build to my releases with a lower min_temp.

gizzburn, 2021-01-22:
Hi Robert I am using a Mk3s with MMU2s and a Bondtech BMG extruder.

MJ.

TheZeroBeast, 2021-01-24:
Ok, so I have added a customer variant for you mate with minimum temp @ 1c, be warned that thermistors are the reason for the 10c setting, I copied the blurb from FW below.

I will also add this to the latest **releases** page and all future releases with have a build for you.
Ok, so I have added a customer variant for you mate with minimum temp @ 1c, be warned that thermistors are the reason for the 10c setting, I copied the blurb from FW below.

I will also add this to the latest **releases** page and all future releases with have a build for you.

[TZB-MK3S_Bondtech-16-EINSy10a-gizzburn-3.2.0-BN2496.hex.zip](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/files/5861997/TZB-MK3S_Bondtech-16-EINSy10a-gizzburn-3.2.0-BN2496.hex.zip)

> // Minimum ambient temperature limit to start triggering MINTEMP errors [C]
> // this value is litlebit higher that real limit, because ambient termistor is on the board and is temperated from it,
> // temperature inside the case is around 31C for ambient temperature 25C, when the printer is powered on long time and idle
> // the real limit is 15C (same as MINTEMP limit), this is because 15C is end of scale for both used thermistors (bed, heater)

closed 2021-01-24 by TheZeroBeast