TheZeroBeast/TZB-MMU2S-Firmware #196: Support for Skelestruder

tmolberg, 2021-01-24:
---
name: Support for Skelestruder
about: Skelestruder seems to be an overall better design x end wise. Using it with singel yields better results in terms of print.
title: ''
labels: enhancement
assignees: ''

---

**Is your feature request related to a problem? Please describe.**
Yes. Modifying the einsy mk3s firmware to accomodate shorter filament path yields "unwatned" results. During load process it seems that the load sequence tend to try and ram the filament way to far resulting in the bondtech gears or stepper stalling due to some short of jam. This can be "fixed" by giving the fillament som help

Modifications done can be seen here; https://www.thingiverse.com/thing:3255143
diff -Naur stock/Firmware/mmu.cpp skelestruder/Firmware/mmu.cpp
--- stock/Firmware/mmu.cpp  2019-06-27 11:32:40.000000000 -0400
+++ skelestruder/Firmware/mmu.cpp   2019-09-01 09:43:50.331743451 -0400
@@ -1462,9 +1462,9 @@
         mmu_command(MmuCmd::C0);
         manage_response(true, true, MMU_LOAD_MOVE);
     }
-    current_position[E_AXIS] += 60;
+    current_position[E_AXIS] += 50;
     plan_buffer_line(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS], MMU_LOAD_FEEDRATE, active_extruder);
-    current_position[E_AXIS] -= 52;
+    current_position[E_AXIS] -= 42;
     plan_buffer_line(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS], MMU_LOAD_FEEDRATE, active_extruder);
     st_synchronize();
 }

**Describe the solution you'd like**
Modifying load sequence to accomodate the shorter filament path, or a clear overview of which parts of the firmware which should be modified to accomodate it. 

**Describe alternatives you've considered**
Alternatives was bashing my head against the wall until I came up with better ideas than asking for help here... 

**Additional context**
See video of filament load sequence. Uploaded one slow motion and one normal. The leverarm shows when the filament enteres the bondtech gears whereas the extrudermotor then reverses the direction of the filament and then starts loading it again. I'm assuming this is intentional for some reason instead of going directly? Or could it be a result of some kinked filament path which the tmc drivers detects so it reverses the direction to try a new reload?  

I've also attached the changes I've done. Minor tweaks with the stallguard homing parameters (using vesconite bearings which tend to give some more binding then regular lm8uu bearings) 

[TZB-MK3S-Firmware-MK3S-TZB-Skelestruder.zip](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/files/5862978/TZB-MK3S-Firmware-MK3S-TZB-Skelestruder.zip)
[Video1.mp4](https://nextcloud.tmolberg.net/index.php/s/gPoWAySo29Dq889)
[Video2.mp4](https://nextcloud.tmolberg.net/index.php/s/LyD2YjmgyJKp4AD)

AbeFM, 2021-01-24:
1) I was using the Skele for at least a year before switching to TZB extruder. Aside from the Skele being cooler since it's open and geared, I didn't find the prusa-based one to be worse. I'd reconsider the extruder switch entirely.
2) Weird loading issues aren't entirely exclusive to the Skele. I tweaked the loading distance on the "stock" TZB extruder because I was always getting "unfixable" jams for no reason. I still think there's something generally wrong. 
3) In the first video, it looks like there's a bump (bad tooth?) on your belt or your driven gear? Probably irrelevant to this but I can't imagine it helps anyway.
4) Do you have a light on the filament sensor, or another way to check it in real time? I believe issues with the sensor for soft materials was what made me go to TZB version. To be fair, it's very touchy with TZB one as well - to get it adjusted right, it sometimes stays triggered even when empty. I'm not sure if anything could be done to up the mechanical "gain" there for a smaller movement to cause a bigger response.

tmolberg, 2021-01-24:
To give some updates, it might not be firmware related but I still find the loading sequence somewhat weird considering its first doing a load, then retract and then load again. Any thoughts on why its doing this or if this is something that could be improved on?

In regards to what I did wrong initially to give some pointers; 
1. Did not have the 1.85 ptfe liner inside the hotend - Very imported for the skelestruder to not create malformed tips when doing unload/load 
2. MMU tention was waaaaay to hard. With it "loosy goosy" its started to prevent grinding but the whole mechanism inside the mmu seems like a grindheaven regardless of what you try to do. 

Response to your questions: 
1. Im not using the skelestruder due to design, im using it strictly because I've had bad experience with the prusa design inside a chamber, with the skelestruder it bumps up the cooling of the hotend just enough for it to be able to print long PLA prints. 
2. I've tweaked the loading distance using the TZB guide for loading distance, that being said my ptfe tube might be waaaay to long for repeatable experience (its around 650mm) 
3. Its just a nick on the sidewall, performance wise it dosent impact the skelestruder at all, if you want I can fix it, but for now I havent used any time on it since its not skipping steps or extruding the correct amount of material.
4. Sorry no, but then again with the skelestruder IR sensor mod im using it seems to be very reliable vs the stock Prusa IR detector which would result in bingo on/off. MMU wise the FINDA seems to be adjusted just correctly. Im not experiencing any issues with detection. All the error messages seems to be valid aswell (i've set up the telegram gateway aswell).

TheZeroBeast, 2021-01-27:
Hi @tmolberg,

If there is demand to have a Skelestruder variant created then I can sure add it, I'll get a build ready for you to test.

@AbeFM 
> 2. Weird loading issues aren't entirely exclusive to the Skele. I tweaked the loading distance on the "stock" TZB extruder because I was always getting "unfixable" jams for no reason. I still think there's something generally wrong.
 - I still think this is a hardware issue and required to catch jams/malformed tips causing missed layers on that colour.
   - Also to note the latest TZB FW has the cutter setting so in this case it would clip the tip and load successfully.