TheZeroBeast/TZB-MMU2S-Firmware #197: Bear Exxa Extruder

fila612, 2021-01-25:
Hello,
just a short question:
i flashed my MK3S with the TZB-MK3S_EINSy10a-3.2.0-BN2495.hex File.
Unfortunately the calibration failed.
So if i want use the TZB MMU Firmware i have to edit the BEAR specific adaptions in the source code, isn't it?
I cannot found the source code for the eons board or ca i use the BEAR Firmware with the TZB MMU Firmware?

thanks

fila612, 2021-01-26:
So, just found SourceCode and changed the Bear parameters.
FW seems to be working, but i think its not usable without a MMU.
Sometimes i start my MK3S without the MMU and then i don't see content in the terminal and when there was some lines, the are "MMU loop, state=5".
So that seems the FW is only usable für non Bear MK3s and only if the MMU are always on.
is that correct?

TheZeroBeast, 2021-01-27:
No need at all for the mmu to be connected, something else will be happening here.

fila612, 2021-01-27:
so for me i found a solution, that a adapt my printer at the z axes to run stock FW on my Bear with EXXA.
Should i close this issue or are you interested to provide also a MK3S variant?

TheZeroBeast, 2021-01-27:
So, just to be clear, you're running [this](https://github.com/gregsaun/bear_extruder_and_x_axis) on a PRUSA MK3S?
I will definingly be adding support, I might even upgrade one of my printers, the X ends look beefy.

fila612, 2021-01-27:
Yes, thats the Bear 2.1 what i‘m Running on my MK3s.
There are two options:
1) adapt the z-top-caps from the bear setup (6mm higher) to run stock firmware
2) add the bear exxa specific Parameters (documented in the Bear firmware) as a Variant of your Version.

frasier2k, 2021-03-23:
Is there any chance that you release an additional variant for this?

TheZeroBeast, 2021-10-02:
I'm working on this now guys, shouldn't take long to create a variant, I'll most likely do a point release when I do.

TheZeroBeast, 2022-01-05:
Hi Guys, I'm sorry but this was missed on my recent releases, I will get it into my next.

TheZeroBeast, 2022-01-05:
Ok, so I have implemented the few changes the Bear calibration FW make in the attached build, could someone please test and confirm please?

[TZB-MK3S-Bear-4.0.2-BN2525.zip](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/files/7811519/TZB-MK3S-Bear-4.0.2-BN2525.zip)

TheZeroBeast, 2022-02-13:
Please review [4.0.2](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/releases/tag/4.0.2) release and re-open if your issue persists.

closed 2022-02-13 by TheZeroBeast