TheZeroBeast/TZB-MMU2S-Firmware #199: Heatbreak Jam Retry Stops Working If Manual Intervention Required

TheClark, 2021-02-04:
I have had this happen twice now.

A heat break jam occurs.  Very nicely this latest firmware retries a few times, which is great.

After a certain number of retries, a manual intervention is required.  

After cutting the tip and hitting retry, the LCD indicates that a retry is happening.  However the stepper motors on the MMU do not move at all.

It looks like some kind of bug where retries are no longer firing correctly, but only after the automatic retries are finished, and a manual intervention is required.

At this point I have to kill the print.  Bug or... ? User error?

Thanks!

TheZeroBeast, 2021-02-05:
Thanks man, I'll try replicate and squash 👍

TheZeroBeast, 2021-02-12:
Hi @TheClark, replicated and fixed the issue with [this](https://github.com/TheZeroBeast/TZB-MK3S-Firmware/commit/4b0e7c792c972a819506b4c0ab556c20a56d6340) commit on the Printer side.

It will be in the next release.

closed 2021-02-12 by TheZeroBeast

TheClark, 2021-02-12:
Love that it is basically a one liner, nice debugging!  Now the work to figure out that one line, that is the tricky part.  Much appreciated.  Cannot wait to check it out.

eurohell, 2021-03-09:
Got a very similar bug: 

When changing filament on the wipe tower, a heatbreak jam occurs. The Extruder moves to a waiting position at the left back corner. When i get it to know, i push the button to retry, hotend starts heating. When about 5°C below target, the extruder moves to the wipe tower without filament loaded, moves down in contact to wipe tower and  instantly moves up and back to waiting position, throwing the jam error to the display again. Pushing the button again repeats this behaviour.

Is this behaviour also covered from your bug fix?