TheZeroBeast/TZB-MMU2S-Firmware #200: LCD Issue - can the firmware force a screen refresh or is there a GCODE that can do that?

Naugrimohtar, 2021-03-03:
My MK3s (+MMU2s) has an issue where the LCD becomes blank occasionally. Sometimes while printing. Other times while just sitting. If I press the button, the display will refresh and be OK for a while. Sometimes this is not OK as is the case when I am being prompted for something, such as to select which filament or if a filament swap was successful. (no OK because if I press the button, I make a selection instead of refreshing the display).

I'd like to fix this issue, I have checked cables and all look ok.

I'd also like a work around, is there a GCODE command I can send to force a refresh of the display?

DRracer, 2021-03-04:
This is more likely an ESD issue on the printer, the MMU doesn't have access to the LCD. Which dialogs do cause problems on your machine?

Naugrimohtar, 2021-03-04:
I suspect you are right it is an ESD issue rather than anything TZB related but I mention it here for several reasons:

You guys know excellent work arounds such as a gcode command maybe I could send via OctoPrint when this happens, or if there is any feature in the firmware that I could leverage to force a full re-draw of the screen as what happens when I click the button.

As far as screens, the issue happens arbitrarily (not ties to being on a screen) but the screens that cause issues for me is when the screen is asking for user input and clicking the button means I don't know what I am clicking. For example -- The screen is working, I send a file that was sliced for 'mmu single' where I need to select a filament, as it starts the screen blanks, I know its asking to select filament but the display is blank. If I press the button, the screen refreshed but then I selected a filament. Does that make sense?

I wish I knew what the actual root cause of the issue is (I have both a MK2.5 and a MK3 and only the MK3 exhibits this issue, but MK2.5 runs stock F/W and no MMU). In the mean time, its a nuisance that I wish I had a work around for. 

One option would be that if the printer is asking for user input, it forcibly redraws the screen, since its unlikely doing anything else at that time?

TheZeroBeast, 2021-03-16:
Interesting, @AbeFM, how long do you have your machine running for this happens?

AbeFM, 2021-03-16:
That's a reasonable question. I'll keep an eye out for it, I know
sometimes, it's definitely static. I feel like sometimes it happens after
not being on very long, but I KNOW it happens after a long time. I also get
scrambled screens (like bad serial, wrong characters showing up but doesn't
look like when it's blank). Generally, getting the screen to refresh
(pressing the button and hoping it's not going to do anything bad) fixes it
from either state.

Naugrimohtar, 2021-03-16:
For me it’s is quite random. Sometimes within minutes of turning on, sometimes hours, sometimes days. Usually while printing, very rarely while just sitting.

AbeFM, 2021-03-16:
I would tend to agree that I have the same experience - but I haven't
really "logged" it. That's my impression, though.

Do you ever get a scrambled screen?

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

AbeFM, 2021-03-16:
https://photos.app.goo.gl/Ln4CGsPY6bnranTJ6 this isn't really what we're talking about, this might be why I got the screen replaced.

Naugrimohtar, 2021-04-01:
I do not get garbled text, just blank.

AbeFM, 2021-04-01:
I suspect I have two issues. :-)

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

TheZeroBeast, 2022-01-05:
Stale issue, closing.
I also believe this isn't related to software and more so the hardware side.

closed 2022-01-05 by TheZeroBeast