TheZeroBeast/TZB-MMU2S-Firmware #201: Unable to Flash MMU2 Firmware

alackofcolor, 2021-03-11:
I've been running a previous version of the TZB-MMU2S firmware for a while now, before deciding to update tonight. Well when I try and flash the firmware in PrusaSlicer, the MMU2 is listed as Arduino Leonardo, instead of as an MMU2. It seems people have had a similar problem here:

https://forum.prusaprinters.org/forum/original-prusa-i3-mmu2s-mmu2-hardware-firmware-and-software-help/no-connection-between-mmu-2.0-and-eindsy/paged/2/

But I've never flashed my MMU2 with Arduino software. So did something happen with the previous version of TZB that I installed via PrusaSlicer that corrupted the bootloader?

TheZeroBeast, 2021-03-16:
Hi @alackofcolor,

Try follow [this](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/wiki/Flashing-Procedure) procedure in the wiki as it works even if previous upload had been interrupted.

alackofcolor, 2021-03-16:
Hello. Thanks for the reply. That seemed to do it. I had noticed that it correctly identified the MMU on startup and then went to Arduino Leonardo after a few seconds. I suppose your instructions to do it under 5 seconds is because of that? I noticed in the Wiki that it says to do the same thing for the printer, but I had no trouble flashing the printer and PrusaSlicer correctly identifies without the 5-second rule. Is there any reason for that? And will I have to flash the MMU like this indefinitely? Or is there a way back to how it's supposed be done according to Prusa?

TheZeroBeast, 2021-03-16:
Hi @alackofcolor, glad this helped.

The 5s for the printer is there to cover for when the previous flash didn't finish/take.

The flashing MMU like this is because I use PlatformIO to build instead of PRUSAs custom Arduino board file.

If you even load stock binary back onto the MMU it will show as MMU after the 5s in PRUSA Slicer.

closed 2021-03-16 by TheZeroBeast