TheZeroBeast/TZB-MMU2S-Firmware #203: Question: Drop in replacement of Prusa MMU firmware?

Redferne, 2021-05-07:
Sorry for daft question, but could really figure out if this firmware can replace the original MMU firmware? Or is it required to flash the custom printer firmware as well? Was hunting around for a custom MMU firmware which has OLED support for the MMU itself. Found this excellent https://github.com/cjbaar/MM-control-01 project, however it is quite outdated and I'm having issues compiling with PlatformIO. Seems like your project has gotten more love :heart:

Naugrimohtar, 2021-05-17:
It both MMU2 and printer firmware as one of the great things is a new communication protocol that helps add a bunch of cool features.

TheZeroBeast, 2021-09-16:
Hi @Redferne, yes this is a drop in for stock hardware as long as the printer and mmu firmwares are updated as they run a different communication protocol as @Naugrimohtar mentioned.

closed 2021-09-16 by TheZeroBeast

lemmingDev, 2021-12-28:
Hi. Can this MMU2S firmware be used with another printer with standard Marlin.

Reason: I've printed and am using the dual filament cutter version and am using it along with TZB firmware on my MK3S Bear, but would also like to use it on another printer with standard Marlin. I was going to create quick disconnects to easily move it. I know the regular MMU2S is supported in Marlin, but wanted to know if your TZB dual cutter version would still work with whatever modifications were added.

Thanks

TheZeroBeast, 2021-12-30:
Hi @lemmingDev,
My FW doesn’t run any stock communication so you would need to re-write marlin to support my communication protocol.

lemmingDev, 2021-12-30:
Thanks for getting back to me

Might have to print the single cutter version

BTW - when the selector is homing to the right, it crashes back and forth for about 30 seconds before working
Do I need to change sensorless homing threshold values in firmware?

TheZeroBeast, 2021-12-30:
Ah I see, as I haven’t setup a dual blade machine yet I don’t have anything to dev with a relies on a community member to test.

I however intent on changing mine to dual and will modify code base accordingly to have it reliable.

lemmingDev, 2021-12-30:
Even with the stock/single blade with the corresponding TZB firmware it crashes back and forth to the right to around 20 or 3 seconds...

gutschke, 2021-12-30:
Just to offer another datapoint. The other day, I upgraded to the dual blade version. Worked great on the first try. And no issues with trying to calibrate itself.

In fact, everything worked so easily, I recommend for others to also upgrade their printer with this mod

TheZeroBeast, 2021-12-30:
Thanks @gutschke, nice, that’s great 👍🎉

@lemmingDev, if you’re getting this homing issue on stock hardware I recommend checking the hardware to ensure it isn’t jamming anywhere and as a last resort you could change stall settings but they aren’t an easy endeavour.

lemmingDev, 2021-12-31:
I have just checked - if I try to help pull the selector to the left just after it touches the right, it works perfectly.
Once homed, it's fine moving left and right 
So - what changes need to be made to the FW to do it without me coaxing it along?
The config.h file seems to have stall guard values and currents

gutschke, 2021-12-31:
I have noticed that the selector can sometimes get hung up, if the blade(s) extend(s) too low or if the groove that the blade(s) pass through is contaminated. That would probably be the first thing I'd check

lemmingDev, 2021-12-31:
Thanks for the tip. 
I will take it apart and remove the blades to see if it helps