TheZeroBeast/TZB-MMU2S-Firmware #204: New Prusa Firmware version 3.10.0

timemaster5, 2021-05-08:
@TheZeroBeast, I remember you said some time ago that notifications didn't work well for you, so this is just a note about the Prusa Firmware version 3.10.0. No rush, and thank you for the great work!

XRyu, 2021-05-20:
I would also love to see that updated @TheZeroBeast  :D Looking forward to this :3

Naugrimohtar, 2021-05-26:
@TheZeroBeast Do you have anything setup for donations to help you with your time and efforts?

gutschke, 2021-08-30:
There is a "Donation" button on https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/releases

I have occasionally sent the author a token of my appreciation, hopefully he was able to get himself all the coffee that is needed to power his amazing work ;-)

TheZeroBeast, 2021-08-30:
I do get myself a coffee from time to time, thank you for your support.

I have been working on rebasing and fixing some bugs but have a hurdle with my development environment not compiling after an update🤦‍♂️

TheZeroBeast, 2021-09-16:
I'm testing a version rebased to PRUSA 3.10.0 now I have my development environment behaving.

TheZeroBeast, 2021-10-02:
Please enjoy, 3.3.0 is out. Any issues please document and post, I'll endeavour to resolve as soon as possible.

closed 2021-10-02 by TheZeroBeast