TheZeroBeast/TZB-MMU2S-Firmware #205: Error on failure recovery

fromanza, 2021-06-06:
Mk3s firmware latest TZB
MMU2s firmware latest TZB
Sliced using latest PrusaSlicer
Using a modded version of TZB extruder

MMU print using extruder 1=PLA and 5=PVA.

The PVA has a high tendency to string so causes frequent blockages but more often is able to continue. However there seems to be a failure mode which results in the print being unable to continue. 

What I see when I get back to the printer (after many hours of printing fine):
-The PVA was completely retracted as expected
-The selector was in its park position all the way to the right (no visible jams or rouge filament)
-The selector stepper motor was off (can turn shaft by hand-also indicting no jams)
-The idler selects filament 1 when left or right buttons are pressed and the filament moved back and forth as expected.
-About 10cm of the PLA filament was sticking out of the MMU (NOT passing thru the selector) which indicates it tried to load again
-I can pull the PLA back into the MMU and then hit the middle button but all this does is rehome the selector and it goes back to its park position and turns off the stepper. Then it churns out another 10cm of the PLA expecting to trigger the filament sensor. 

So to me it seems the code is not telling the selector to go to the active/next filament. If the selector was in the right place I assume the filament would have hit the filament sensor and the filament would then have moved to the extruder and the print continue. 

All of this happened a few times but close to 50 material changes into the print so I know the hardware is fine 🧐

TheZeroBeast, 2021-06-16:
Hi @fromanza,

Just for historic purposes, the current version of my printer and MMU firmware is 3.2.0.

I'm almost positive this is a bug I have already found and am currently testing the fix, I'll double check in the near future and do a release.

fromanza, 2021-06-17:
Great! And sorry I should have mentioned the actual firmwares I’m on, will do so in the future but glad you’ve picked something up. Just for interest sake I went back to the stock prusa firmware and OMG it is terrible… I forgot how bad it was so thank you for your efforts, without which my MMU would be in the bin.

TheZeroBeast, 2021-06-17:
You’re welcome, glad it’s helped, it’s saved my sanity as well 😂

Pun-e, 2021-06-24:
Hi there, I'm not sure if this is the same issue or not, but whenever the MMU fails to the point it needs a manual intervention, I cannot resume the print.

Selecting resume seems like it's going to work, printer moves to the last position, and then it just returns to the error state.

Octoprint just shows that the printer is not responding, and after this happened (along with the other issue) I made the decision to move back to the official firmware. Unfortunate, as your work seems much better (for real how did you make it so quiet) but I can't afford for literally a single MMU failure to jettison the whole print.

3.2.0 on both MMU and MK25S_Rambo fwiw

eurohell, 2021-06-25:
> Hi there, I'm not sure if this is the same issue or not, but whenever the MMU fails to the point it needs a manual intervention, I cannot resume the print.
>
> Selecting resume seems like it's going to [...]

Had a similar issue a few weeks ago, and Robert answered that this behaviour will be fixed in the upcoming iteration of TZB Firmware. [https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/issues/199#issuecomment-794073516](url)

Marek125, 2021-07-09:
This is the only mistake I experience with this FW, I don't print almost anything on the official FW, it's in a deplorable state.  TZB best FW for MMU I'm waiting for a new FW where it will be missing repaired as for soup salt thank you for your work :)

TheZeroBeast, 2021-09-16:
Sorry for the late update guys, I've been having compilation isues on my development environment which have now been fixed. I'm testing a version rebased to PRUSA 3.10.0 with this fix that I'll release once working properly.

TheZeroBeast, 2021-10-02:
Please enjoy, 3.3.0 is out. Any issues please document and post, I'll endeavour to resolve as soon as possible.

closed 2021-10-02 by TheZeroBeast

Marek125, 2021-10-03:
Hello, the new build of TZB 3.3.0 does not allow me to choose the
experimental functions of FW for Alternative Fan. I use TZB 3.3.0 Bondtech
FW.

https://help.prusa3d.com/en/article/selected-third-party-components-mk3s_161213

Thanks Marek