TheZeroBeast/TZB-MMU2S-Firmware #206: MMU Doesn't retract filament after a filament eject

Pun-e, 2021-06-16:
**TZB-MMU2S Release Version**
3.2.0

**Describe the bug**
MMU Doesn't retract filament after a filament eject

**To Reproduce**
Steps to reproduce the behavior:
1. Eject any filament
2. Click ok
3. MMU moves over ejected filament **without retracting it**

**Expected behavior**
The MMU retracts the filament before re-loading it - like the stock firmware does

I'm glad that I can now eject filament without preheating the nozzle, but I would rather have to wait for the hotend to heat up than remember I need to manually feed the filament back. 

If manually feeding the filament was a viable option then why would I be bothering with an automated eject sequence?

Naugrimohtar, 2021-06-16:
I am also curious why the eject filament feature doesn’t retract once done. My main is is to check tips, snip them. Would be awesome if it then retracted it back.

TheZeroBeast, 2021-06-16:
Hi guys, sorry I haven't got this in the wiki yet but the right button on the MMU does the retract, left does feed more.
This is to give some control back to the user, being able to feed or retract more when you have pulled it out further for example.

Let me know if you still have questions about this.

closed 2021-06-16 by TheZeroBeast