TheZeroBeast/TZB-MMU2S-Firmware #207: Bowden Length Calibration

Ondak87, 2021-07-07:
Printer type - MK3S+
Printer firmware version - TZB-MK3S_EINSy10a-3.2.0-BN2495
MMU2S
MMU upgrade firmware version - TZB-MMU2S-3.2.0-BN371
extruder - TZB MK3S MMU2S (No Tower)

Describe the bug
When bowden length calibration is new value doesnt change position ,,1,, bowden length calibration are change only for ,,2-5,,.

Expected behavior
Bowden Length Calibration are change on new value for ,,1-5,,.

TheZeroBeast, 2021-09-16:
Hi @Ondak87, can you please provide more details and all the steps you came to your conclusion so I can attempt to troubleshoot?

Ondak87, 2021-09-17:
Hello @TheZeroBeast, i'm sorry, I did a new bowden calibration test  and problem was in MMU board.

Tiket you can closed

closed 2021-09-17 by Ondak87