TheZeroBeast/TZB-MMU2S-Firmware #209: Check position mmu

Sterbo88, 2021-09-12:
hi, I have the latest firmware version 3.2.0 and I noticed that after a while I don't use the printer, when I start a print it still resets the mmu position ... during a mmu print, it did it about 2 times without apparent reason and without problems continued ... is this normal?

TheZeroBeast, 2021-09-13:
Hi @Sterbo88,

After a short period of inactivity the motors all turn off to be quiet and same some power. Once you start commanding it again it needs to check its position.

closed 2021-09-13 by TheZeroBeast

Sterbo88, 2021-09-13:
> needs

why did he do it even during a mmu print?

TheZeroBeast, 2021-09-15:
The MMU will rehome once every 100 tool changes as well to ensure it doesn't get out of sync which can be an issue on large MMU prints.