TheZeroBeast/TZB-MMU2S-Firmware #210: Firmware Addition.

gizzburn, 2021-09-13:
Hi Robert you have done my Gizzburn firmware version and I am very grateful, but I have another request please. I have changed my MMU2 selector to the two blade cutter version and I need these changes in the firmware could you add these for please so I can use it.

https://www.prusaprinters.org/prints/16666

On "mottion.cpp" line :35

//const int selectorStepPositionsFromHome[EXTRUDERS+2] = {-3700, -3002, -2305, -1607, -910, -100, 0}; //original part

const int selectorStepPositionsFromHome[EXTRUDERS+2] = {-3178, -2479, -1782, -1085, -387, -27, 0} ; // this part

Gizzburn.

TheZeroBeast, 2021-09-15:
Sure can, I just got my dev environment compiling again so release soon after testing.

TheZeroBeast, 2021-10-02:
Please enjoy, 3.3.0 is out. Any issues please document and post, I'll endeavour to resolve as soon as possible.

closed 2021-10-02 by TheZeroBeast