TheZeroBeast/TZB-MMU2S-Firmware #211: Persistant tune temperature settings per filament

SomeoneElse58, 2021-09-27:
**Is your feature request related to a problem? Please describe.**
Tuning temp through the LCD menu works great when it's a single filament print but doesn't quite work as one would hope when doing an MMU print.

Example: I'm running a 5 filament MMU print. The temperature of filament 3 (gray) _needs_ be bumped 10*. While gray is being extruded I use the LCD/button, go into tune, change nozzle temp +10*. It works, but only until the filament is swapped. Any time gray gets extruded afterwards the temp is reverted to whatever setting it was at when sliced.

**Describe the solution you'd like**
Is there a way to get the temperature setting from the tune menu to stay persistent throughout an MMU print for that specific filament currently in use? 

Sticking with the example: Can you make it so that +10* change is applied whenever gray is being extruded from that point on? So the temp doesn't revert once the filament is swapped out.

**Describe alternatives you've considered**
Only two real alternatives at the moment. One, manually tune the temp every time that filament is used - unrealistic. Two, stop the print, fix the temp in the slicer, and then reprint from zero.

TheZeroBeast, 2021-10-02:
Hi @SomeoneElse58,

As you have hinted at you would need to maintain your filament profiles in your slicer so you have the correct values.

I also understand your issue and will see if I can incorporate it into the firmware at some stage in the future.

Ondak87, 2021-10-04:
Hello @SomeoneElse58 

Temperature setting for each train can be set in the slicer.

I personally recommend downloading a SuperSlicer where you can even set the temperature when the train is change

SomeoneElse58, 2021-10-04:
Hi @Ondak87

I'm aware of slicer settings. The point of my request is for instances when you need to change a temperature mid-print and want that changed temperature to stay at that modified temperature for the rest of the print. 

When printing with a single-filament there is no problem. However, any filament temperature changes made through the Tune settings on the LCD menu of the printer are reverted to whatever temperature amount it was set to during slicing once that filament is swapped out during an MMU-print. I am hoping there is a way to remedy this so that Tune changes don't revert for the duration of the print.

I'm not asking for the printer to change the g-code on the SD card. But I'm just a layman, so I can't make solution recommendations.