TheZeroBeast/TZB-MMU2S-Firmware #212: MMU2 Filment Idler Banging/Grinding

Darkknight33, 2021-10-02:
**TZB-MMU2S Release Version**
3.2.0

**Describe the bug**
Idler rotates with excessive force to (I think) clockwise limit during startup and calibration. It's much louder and more dramatic than with stock firmware. I flashed the TZB MMU2 firmware a second time in case there was problem, but still the same result. It's overdriven with a sound like during Z calibration when it's finding the upper limit -- okay for calibration, but I don't want to hear that grind every time it starts up. 

**To Reproduce**
Steps to reproduce the behavior:
Flash TZB MK3S_Einsy10a_3.2.0 Firmware
Flash TZB MMU2S-3.2.0 firmware
Reset

**Expected behavior**
The stock firmware startup is pretty quiet, basically a couple quiet clicks. MMU2 should have normal limit finding, does not apply so much motor current or overdrive when finding the limit.

TheZeroBeast, 2021-10-02:
Hello @Darkknight33,

the idler stallguard values have been tuned for the correct spring tension which could be a factor in your setup loosing steps instead of triggering stallguard.

Can you advise if you’re using original Prussia supplied board and steppers?

Darkknight33, 2021-10-02:
Yeah, it’s all original Prusa parts from a MK3S+ purchased at the end of last year, and an MMU2S purchased in March. 

I printed and installed your extruder upgrade. I’ve made a few small prints, and it prints okay, just occasionally pops up with that error from the other issue when it’s idling. 

The tension wasn’t changed between firmwares. The screws are roughly 1mm or so below the surface level of the MMU2. I don’t remember it grinding like that, even when the idler was completely open — but it’s possible I never reset it in that state. 

I didn’t want to over tighten it, but when the print stops next, I can apply more pressure. 

What pull tension should it be set at?

TheZeroBeast, 2021-10-02:
Hmm, thank you for the info.

It could just be stepper impedance variances but if so I'll show you which values to tune for your setup.

First, this 1mm below surface, try two complete turns from surface.

Darkknight33, 2021-10-04:
The new firmware has not resolved the grinding. I have tightened the screws per your instructions, it's still pretty loud. I tried different positions, and I do not think it is a screw torque issue. I think it's just over driving the idler too much. 

I wonder if somehow we have different MMU hardware? Maybe they changed the stepper motors?

TheZeroBeast, 2021-10-04:
If so you will need to tune the values.
Have you got a development environment setup and compiling already?
If not please get setup as per the read me and I’ll find the lines you’ll have to adjust until you have a proper stallguard detection.

Darkknight33, 2021-10-04:
I do not have the IDE setup, but I will work on this week. Thanks for the help.

TheZeroBeast, 2021-10-04:
Sounds good, platformIO is what I dev in which is easy as to setup.

eurohell, 2021-10-05:
I got this bug also many times on different printers, the problem is that the 2 screws that fix the idler to the motor are too loose (but be careful, idler can be easily cracked when overtightened, perhaps it already has cracked?). So tighten these 2 screws a little bit or check if idler is cracked, then print a new one. The problem will disappear, because for now, the motor rotates freely in the idler.

TheZeroBeast, 2022-01-05:
Stale issue, closing.
@Darkknight33, once you have a dev. environment up and working post back and I can re-open.

closed 2022-01-05 by TheZeroBeast

Darkknight33, 2022-01-05:
Sorry for forgetting about this. My MMU2S unit is not cracked, and works without issue on the stock firmware. Tightening the screws to nearly the point of binding the tension springs did not resolve the motor grinding. I believe the drive current is just set too high for my selection stepper in the TZB firmware. 

I spent my time working on other issues in the intervening time. Between the new extruder, the 5 filament rewinders and discovering and correcting how completely inaccurate the temps are, it prints somewhat reliably now. I don’t want to mess with config by experimenting with different drive currents to solve an otherwise non-existent problem. The only print problems I have now are related to the filament unload speed. Sometimes the filament jumps the spool when the unload speed is faster than the reaction time of the return spring on a heavy/full spool. I printed for roughly 4 days straight with only a couple jams at the end related to that problem.

rogery555, 2022-08-13:
Same issue for me, idler grinds terribly when booting the MMU  4.0.2 firmware