TheZeroBeast/TZB-MMU2S-Firmware #213: Error: Static values overwritten on LCD when using Octoprint

Darkknight33, 2021-10-02:
**TZB-MMU2S Release Version**
3.2.0

**Describe the bug**
When using TZB MK3S Einsy10a 3.2.0 firmware with Octoprint, an error message "Error: Static values were overwritten" will display on the LCD, and you can do nothing but reset the printer. After resetting, it works fine with octoprint connected.

This is using an RPI 4, 4GB with Octoprint 1.6.1 w/ OctoDash installed

**To Reproduce**
Steps to reproduce the behavior:
1. Install TZB firmware
2. Start Printer
3. PI boots slower than printer, causes reset of printer interface after it boots (normal behavior)
4. After reset, instead of normal info screen, an error message is displayed on the printer LCD that cannot be recovered from except from a manual reset.

**Expected behavior**
Printer should reboot normally after Octoprint finishes starting up

TheZeroBeast, 2021-10-02:
Hi @Darkknight33,

I have done a fair bit on this issue in my latest code that I’m about to release as 3.3.0.
We will have to see if you still get this with that build.

Darkknight33, 2021-10-02:
OK, when it releases soon ™️, I’ll test it out.

TheZeroBeast, 2021-10-02:
Ok, dig in @Darkknight33, it's live 😉

Darkknight33, 2021-10-03:
I just started an 8hr print. I’ll hit it tomorrow and let you know. Thanks.

Darkknight33, 2021-10-03:
Ok, a more serious problem. After flashing the 3.3.0 firmware on both the MMU2S then the printer, the printer does not recognize that MMU2S is even connected. It's behaving as if it's not there at all. Which is to say, there is no reference in the sensors page for the sensor in the MMU2, no option to select which filament to load, and when trying to load a filament, the printer is expecting it from the extruder. The MMU2 lights are off, and it does not cycle through the startup routine.

The flash went ok, but here's the Flash log:

> Flashing Original Prusa MMU 2.0 Control, looking for VID/PID 0x2c99/3 or 0x2c99/4 ...
> The Original Prusa MMU 2.0 Control device was not found.
> Found VID/PID 0x2c99/3 at `/dev/cu.usbmodem141301`, flashing ...
> avrdude-slic3r -v -p atmega32u4 -c avr109 -P /dev/cu.usbmodem141301 -b 57600 -D -U flash:w:0:/Users/darkknight/Downloads/TZB-MMU2S-3.3.0-BN374.hex:i
> 
> avrdude-slic3r: Version 6.3-20160220-prusa3d, compiled on Jan 11 2021 at 14:04:11
>                 Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
>                 Copyright (c) 2007-2014 Joerg Wunsch
> 
> 
>                 Using Port                    : /dev/cu.usbmodem141301
>                 Using Programmer              : avr109
>                 Overriding Baud Rate          : 57600
>                 AVR Part                      : ATmega32U4
>                 Chip Erase delay              : 9000 us
>                 PAGEL                         : PD7
>                 BS2                           : PA0
>                 RESET disposition             : dedicated
>                 RETRY pulse                   : SCK
>                 serial program mode           : yes
>                 parallel program mode         : yes
>                 Timeout                       : 200
>                 StabDelay                     : 100
>                 CmdexeDelay                   : 25
>                 SyncLoops                     : 32
>                 ByteDelay                     : 0
>                 PollIndex                     : 3
>                 PollValue                     : 0x53
>                 Memory Detail                 :
> 
>                                          Block Poll               Page                       Polled
>                   Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
>                   ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
>                   eeprom        65    20     4    0 no       1024    4      0  9000  9000 0x00 0x00
>                   flash         65     6   128    0 yes     32768  128    256  4500  4500 0x00 0x00
>                   lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
>                   hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
>                   efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
>                   lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
>                   calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
>                   signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
> 
>                 Programmer Type : butterfly
>                 Description     : Atmel AppNote AVR109 Boot Loader
> 
> Connecting to programmer: Found programmer: Id = "CATERIN"; type = S
>     Software Version = 1.0; No Hardware Version given.
> Programmer supports auto addr increment.
> Programmer supports buffered memory access with buffersize=128 bytes.
> 
> Programmer supports the following devices:
>     Device code: 0x44
> 
> avrdude-slic3r: devcode selected: 0x44
> avrdude-slic3r: AVR device initialized and ready to accept instructions
> 
> Reading | ################################################## | 100% 0.00s
> 
> avrdude-slic3r: Device signature = 0x1e9587 (probably m32u4)
> avrdude-slic3r: reading input file "/Users/darkknight/Downloads/TZB-MMU2S-3.3.0-BN374.hex"
> avrdude-slic3r: writing flash (17412 bytes):
> 
> Writing | ################################################## | 100% 1.39s
> 
> avrdude-slic3r: 17412 bytes of flash written
> avrdude-slic3r: verifying flash memory against /Users/darkknight/Downloads/TZB-MMU2S-3.3.0-BN374.hex:
[... 56 lines cut ...]
>                 Vtarget         : 0.0 V
>                 SCK period      : 0.1 us
> 
> avrdude-slic3r: AVR device initialized and ready to accept instructions
> 
> Reading | ################################################## | 100% 0.01s
> 
> avrdude-slic3r: Device signature = 0x1e9801 (probably m2560)
> avrdude-slic3r: reading input file "/Users/darkknight/Downloads/TZB-MK3S_EINSy10a-3.3.0-BN2520.hex"
> avrdude-slic3r: writing flash (230474 bytes):
> 
> Writing | ################################################## | 100% 40.67s
> 
> avrdude-slic3r: 230474 bytes of flash written
> avrdude-slic3r: verifying flash memory against /Users/darkknight/Downloads/TZB-MK3S_EINSy10a-3.3.0-BN2520.hex:
> avrdude-slic3r: load data flash data from input file /Users/darkknight/Downloads/TZB-MK3S_EINSy10a-3.3.0-BN2520.hex:
> avrdude-slic3r: input file /Users/darkknight/Downloads/TZB-MK3S_EINSy10a-3.3.0-BN2520.hex contains 230474 bytes
> avrdude-slic3r: reading on-chip flash data:
> 
> Reading | ################################################## | 100% 29.53s
> 
> avrdude-slic3r: verifying ...
> avrdude-slic3r: 230474 bytes of flash verified
> 
> avrdude-slic3r done.  Thank you.

TheZeroBeast, 2021-10-03:
Cool, I’ll see if I can work out what’s going on 👍

TheZeroBeast, 2021-10-03:
Ok, my hex builds stuffed up on my final build 🤦‍♂️
I have updated the MMU2S hex files on the release page.
Thank you for the heads up.

Darkknight33, 2021-10-04:
The new firmware seems to have resolved this issue, as well as the MMU not working at all. This issue seems to have been solved, but issue #212 is still very prevalent. I will close this and comment over there about it.

closed 2021-10-04 by Darkknight33