TheZeroBeast/TZB-MMU2S-Firmware #220: Automatically unload and retract filament before attempting to cut

gutschke, 2021-12-28:
This is the first time I have actually tried the "cut filament" option from the menu. So, no idea whether this is a bug in the TZB firmware or if this is just the same behavior as in the stock firmware.

If no filament is currently loaded, cutting works perfectly. The MMU2S produces beautiful ~1cm cuttings, and the filament has a clean tip that loads easily. But if I accidentally ask the printer to cut filament while it is still loaded into the extruder, it makes a futile attempt to move the blade. And afterwards the filament is fouled up in the MMU2S. I had to actually disassemble it and loosen the selector in order to clear the error condition.

In an ideal scenario, the printer would detect this situation, realize that the blade can't move until the filament is unloaded, and thus offer to unload first. But that requires preheating to the correct nozzle temperature. So, there are a couple of user interactions required before the MMU2S is ready. Maybe, it's easier to simply show an error message whenever the filament sensor still shows the filament as loaded when the user requests it to be cut. On the other hand, if the sensor in the extruder shows empty, and the FINDA shows loaded, then that's presumably a condition that can be cleared automatically.

N.B.: I installed the dual-blade selector and the matching firmware. I don't think this makes any difference for this particular feature, but just wanted to mention it.

TheZeroBeast, 2022-01-05:
Hi @gutschke,

This is a good find, I haven't tried this before and it should be fixed in software. I'll get it sorted.

TheZeroBeast, 2022-02-10:
Fixed in the upcoming release by advising filament loaded and doesn't cut.
I will probably improve it in the future and have it prompt to heat/unload then cut.

closed 2022-02-10 by TheZeroBeast

TheZeroBeast, 2022-02-12:
Scratch that, I coded it to preheat if loaded then cut the loaded channel. Once this is done you could go into the menu and cut other channels.