TheZeroBeast/TZB-MMU2S-Firmware #223: MMU2S does not continue print after unload failure

itscountolaf, 2022-01-10:
Ensure you're running the latest **RELEASE** and you've checked the **Wiki** for answers before reporting. Any issues reported regarding previous **RELEASES** will automatically be closed and not responded to, this ensures the developer maintains focus & streamlines the development process.

**TZB-MMU2S Release Version**
4.0.1

**Describe the bug**
MMU2S had an unload failure. After fixing it my MK3S says "MMU needs user attention" and stays at 195 degrees.
The middle button on the MMU loads the filament as usual but does not continue to load it to the nozzle and resume the print.
Instead the middle button loads filament, left button moves selector to the left, right button moves selector to the right.

**To Reproduce**
Steps to reproduce the behavior:
1. Get an unload failure
2. Press the MK3S knob
3. Fix unload error on MMU2S (there is no filament in FINDA, FINDA LED is red)
4. See error

**Expected behavior**
The MMU2S should load the filament to the nozzle and the whole print should resume.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Serial Log**
Please attach and serial logs of this issue.

**Additional context**
Add any other context about the problem here.

TheZeroBeast, 2022-02-01:
When you say the MMU middle button attempts to load filament do you mean it loads then parts as if it doesn't know its printing and in an error state? Just trying to understand what is going on so we can sort the issue out.

TheZeroBeast, 2022-02-02:
Hi @itscountolaf, thank you for reporting this, I think it is related to #219 which I will be testing a fix for in the coming days.

itscountolaf, 2022-02-02:
Exactly. Imagine the printer says it is in an error state and the MMU does not know this and acts as if you just powered it on.

TheZeroBeast, 2022-02-06:
Hi @itscountolaf, it sounds like your MMU2S rebooted when fixing the issue which gets it out of sync with the printer.

This has been on my todo list for a couple years so I decided to get it sorted.
I have coded the printer to recover the connection/sync if it sees the MMU2S booting if it had been enabled priviously.

So I have tested and have it re-syncing while idle, printing and mid MMU2S error recovery so if you have it open and bump the power on the MMU2S it will recover and let you continue your print.

itscountolaf, 2022-02-07:
That sounds plausible. When the issue occurred, I just reset the MMU with the button on the side but it did not get in sync. But I did not power cycle the MMU.

I will try it the next time the problem occurs. Fortunately it didn't happen again since opened up this issue.

Other than that the firmware and the MMU work like a charm.

TheZeroBeast, 2022-02-13:
Please review [4.0.2](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/releases/tag/4.0.2) release and re-open if your issue persists.

closed 2022-02-13 by TheZeroBeast