TheZeroBeast/TZB-MMU2S-Firmware #225: Handle mmu reset/power loss event.

sean97140, 2022-02-21:
**Is your feature request related to a problem? Please describe.**
If I were to reset the mmu, or pull the power to it during a failed load state. It will not recover from this. The stock firmware is capable of recovering from this. 

**Describe the solution you'd like**
I would like a "reset mmu state" menu option to get the mmu back in working state if I were to need to disconnect it for a tear down during a super long print. 

**Describe alternatives you've considered**
The only way to recover from this state is to make a power loss event happen. Unplug the printer for a few seconds, plug it back in, and let it recover. The mmu will then reset and things will continue normally. 

**Additional context**
I had a print with a few thousand color changes, and the mmu needed servicing mid-print. I disconnected it to clean it and replace a part that had failed and upon plugging it back in. It would not resume the failed load. I tried everything...then ended up pulling the plug as a last-ditch effort which got the mmu to resume correctly.

eurohell, 2022-03-08:
Got a similar issue, what worked after reconnecting the mmu is to go into the service menu by moving selector to the virtual 6th slot, on the right side, pressing middle button, wait for 2 rightmost blinking leds, red and green, then pressing middle button. you are in the service menu. then go to the leftmost position (press left button 4 times) and then the middle button to leave service menu. after doing this, mmu2 board and einsy rambo are communicating again. This always works for me.

TheZeroBeast, 2022-03-13:
This feature was implemented in the latest firmware. Please review and if you have an issue with that we can re-open.

closed 2022-03-13 by TheZeroBeast