TheZeroBeast/TZB-MMU2S-Firmware #226: Toggle option for extruder movement on MMU error

SomeoneElse58, 2022-03-04:
Please add an option to toggle whether or not to move the extruder when an MMU error happens and the hotend cools off.

Or change it so that the extruder moves to a safe z-height before moving in the x/y directions.

[=(](https://clips.twitch.tv/RockyCrunchyOctopusKAPOW-lDczyqUjfEFydT9V)

I originally posted this on the PrusaSlicer section of github but was told it was a firmware issue.

Cheers

TheZeroBeast, 2022-03-13:
Hi @BaconFase, you would need to have it move and it should be raising in the Z direction from memory. In any case I haven't modified this from the base prusa print FW.

I will need to test if it raises and can add that if needed.

BaconFase, 2022-03-14:
I assume it was because I was using the experimental 'no sparse layers' setting that it didn't raise the Z high _enough_ for it to clear the actual model itself, but it does raise up from the purge tower a little.