TheZeroBeast/TZB-MMU2S-Firmware #227: Force filament cut on each filament change

gutschke, 2022-03-13:
I was having trouble with my hot-end (prints fine in PLA/PETG but always causes thermal run-away for any higher temp than that). I figured, either the heater cartridge or the heater block were defective, but I had already changed those out before. So, on a whim I bought a Revo 6. Love it. Great concept. Incidentally, it seems to have fixed my thermal run-away; so that's good.

But enough of the backstory, with the Revo 6, the MMU2S doesn't always work reliably, as the ramming settings aren't right. This was expected, and I knew that going into it. That's why Prusa had a custom-heatbreak for their MMU2S. I am sure, eventually there'll be official work-arounds to optimize this behavior. But in the meantime, the most pragmatic solution would be to cut the filament on each filament change. That way, there always is a clean tip with no residual stringing.

I suspect there is some poorly documented way that I can do this with custom G-code, but it would be much nicer if I could just switch a flag either in the printer's configuration menu or in the "hidden" menu for the MMU2S. Is this something you would consider adding? I know I am at the cutting edge here, so I deserve what I am getting :-) But the Revo 6 is just too tempting. It really is a big improvement over the previous hot end. Honestly, I'd recommend you pick one up and play with it.

ursius2020, 2022-04-18:
By looking at the code it seams easy. Just add "mmctl_cut_filament(new_extruder);" before "load_filament_withSensor();" in theorie. Finishing my print in a few hours and starting to test my theorie

ursius2020, 2022-04-18:
Alterenatively there is following option in the printer Firmware:
`// This is experimental feature requested by our test department.
// There is no known use for ordinary user. If enabled by this macro
// and enabled from printer menu (not enabled by default). It cuts filament
// every time when switching filament from gcode. MMU_HAS_CUTTER needs to be
// defined.

//#define MMU_ALWAYS_CUT`
No need to change the Firmware of the MMU