TheZeroBeast/TZB-MMU2S-Firmware #229: MMU does not do anything in mid-print

eurohell, 2022-04-24:
**TZB-MMU2S Release Version**
MMU2S-TZB 4.0.2

**Describe the bug**
Several times on different printers, in mid-print on a multi-color print the MMU suddenly does nothing. Sometimes there are no LEDs lighting, only the inner ones on the MMU board, and sometimes many of them are lighting, as in the picture attached. Always, the left motor is very hot (around 65°C, while on running printers this one is only 49°C warm) and the mmu does not react to any pressed button on it. On earlier releases of your software, you could reboot the MMU by pressing the reset-button, then go into MMU menu and leave it, and the communication between EINSy and MMU board has been recovered. If you do this now, the printer suddenly goes on with the print, but with cold extruder (0°C), and according to this with no extrusion, and exact 20mm above the right layer, and follows the path it should print. You can do nothing against this problem, the print is messed up. I do not know if this behaviour came with 4.0.1 -> 4.0.2 or before, but the versions about half year ago haven't had any of these problems. I think it is no hardware error, because it happened on at least 5 different printers on our farm.

**To Reproduce**
Simply print a multi-color-print and wait, the error will happen.

**Screenshots**
[image]

TheZeroBeast, 2022-05-01:
This is not great. Sorry to hear this is happening @eurohell.
I haven’t had any errors like you’re describing, if I come across it I will try resolve 👍

Clank50AE, 2022-06-05:
I too just experienced this issues just now. Simple 2 color print with 51 changes. About an hour into the print, the MMU2S did nothing. My Mk3s+ acted mid print that it was doing a simple filament change, brought the extruder up about 100mm, loaded the filament like normal, asked if it was loaded then returned to print. It would purge on the tower, unload then reload again with the same filament and printed all parts with the same color. No Idea why. Thought the MMU froze but it was responding to button presses

eurohell, 2022-06-06:
It drove me crazy to have that issue on many of our 20+ machines more than once a day, so the only solution that works for me is reverting to 3.3.0, which runs flawlessly (but has the z-lift-bug) and you can recover from any error state if you know how until the new versions run stable.

Clank50AE, 2022-06-06:
I'm beginning to wonder if this has something to do with the FINDA on the MMU. My machine kept trying to swap rolls of this blue filament but is fine with the black. I adjusted the FINDA by removing filament from the selector, lowering the FINDA until it triggered and backed it off just until it read 0. So far, it's printing fine. 

My thought is that the FINDA is really sensitive in this firmware somehow, reads the filament is just a hair smaller than 1.75 and thinks it ran out. During a multi-color print, I assume it just thinks it ran out on that particular slot so it sticks with the one color it has and keeps printing. I'm probably wrong in this assessment but I'm going to try a multi color print again later and ill comment back my findings

TheZeroBeast, 2022-07-14:
#228 looksnto be related. Has anyone looked into the code for the issue??