TheZeroBeast/TZB-MMU2S-Firmware #230: Which Firmware Hex File to use for MMU2S with stock single blade?

talldonkey, 2022-04-25:
**TZB-MMU2S Release Version**
https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/releases/tag/4.0.2

**Describe the bug**
I am not sure which MMU2S Hex file to load into the MMU2S when using the stock Prusa Single Blade selector.  In 4.0.1, it was clear as you had a MMU2S hex file named "[TZB-MMU2S-4.0.1-BN376.hex](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/releases/download/4.0.1/TZB-MMU2S-4.0.1-BN376.hex)"

But now in 4.0.2, you have 

[TZB-MMU2S-DoubleCutter-4.0.2-BN380.hex](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/releases/download/4.0.2/TZB-MMU2S-DoubleCutter-4.0.2-BN380.hex)

and

[TZB-MMU2S-TZBDualBlade-4.0.2-BN380.hex](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/releases/download/4.0.2/TZB-MMU2S-TZBDualBlade-4.0.2-BN380.hex)

Maybe I missed it, but I could not find it documented or named which file to use with the stock single cutter.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/releases/tag/4.0.2
2. Inspect Assets, look for something that is named MMU2S single / stock etc

**Expected behavior**
More clarity on which hex file to use with stock selectors.

Tony763, 2022-04-25:
Hi @talldonkey, try to use TZBDualBlade. At Release page, it is listed as one:

MMU2S Stock/Single Blade/[TZBDualBlade](https://www.prusaprinters.org/prints/133891-mmu2s-tzb2-dual-blade-filament-path-refinement-mod)
MMU2S Dual Blade see [here](https://www.prusaprinters.org/prints/16666)

ackerthehacker2, 2022-05-01:
On a side note, do you have to upgrade the MKS firmware if you all you want is the service position in the MMU to disable it?

TheZeroBeast, 2022-05-01:
DualBlade one tagged hex is also for stock setup 👍

As this FW is completely different to stock both printer and mmu unit need flashing to function.

closed 2022-05-01 by TheZeroBeast