TheZeroBeast/TZB-MMU2S-Firmware #233: Readme.md file requires Update

bryn51, 2022-07-14:
I can happily build this code using Platformio, within VS Code, on my Mac.  I am an experienced coder in C and C++

But the instructions found in readme.md make no sense at all to me.  They appear to be incomplete or imprecise.  

Here are my own Build instructions:

1. download the code base zip file and open it in your file system.  This creates a folder containing src folder.  Platform.ini is found at the base.
2. Start VS Code, and open the folder that was created when you 
3. Open the project at the base folder that contains platform.ini
4. click on the platformio button (little face icon).
5. Select either the MMU2S-TZBDualBlade option or the MMU2S-DoubleCutter option, depending on which hardware you already printed and installed.  Open the 'General' menu option
6. Click the build button in that option.
7. The Terminal window opens on the main screen and you see progress of building which happens pretty quick.
8. it should say at the end:

RAM:   [==        ]  15.5% (used 398 bytes from 2560 bytes)
Flash: [======    ]  61.9% (used 17754 bytes from 28672 bytes)
Building .pio/build/MMU2S-TZBDualBlade/TZB-MMU2S-TZBDualBlade-4.0.2-BN380.hex
 [SUCCESS] Took 1.02 seconds 

The file TZB-MMU2S-TZBDualBlade-4.0.2-BN380.hex constitutes the operational firmware code that needs to be installed into the MMU MCU device, which can be done using PrusaSlicer with the MMU plugged into an available USB-A port on your computer.  This means that if you have a MacBook you are going to need a dongle to provide USB-C to USB-A conversion.

Thats OK so far as it goes, but whats not clear are the instructions :

The HEX which is placed within the .pio root folder still requires the addition of the the bellow line.
 ~~~
 ; device = mm-control
 ~~~

Now the Hex file (TZB-MMU2S-TZBDualBlade-4.0.2-BN380.hex) looks like this (at the start):
`:100000000C94CE000C94F6000C94F6000C94F600C0
:100010000C94F6000C94F6000C94F6000C94F60088
:100020000C94F6000C94F6000C9440140C94AF144D
:100030000C94F6000C94F6000C94F6000C94F60068
:100040000C94F6000C94F6000C94F6000C94F60058`

So that instruction looks like it is the wrong kind of thing to be adding to the hex file.  
- It is text not hex.  And it looks like a comment.  What is the purpose of this addition to the file ?
- Its unclear if the new line of text ought to be placed at the start or the end of the file.

I do not have confidence in this.
Revised instructions are needed.  The readme file needs to be updated with more details somewhat similar to what I have given above, to guide less experienced people that may not be confident about building code files.

TheZeroBeast, 2022-07-14:
The instructions are fine, I don’t agree they need to be so detailed.

The note about the additional line is for use uploading via Prusa Slicer.

closed 2022-07-14 by TheZeroBeast

bryn51, 2022-07-14:
Yeah OK.

But you are not the one have to deal with the upgrade.
I am an experienced maker and coder, but still found myself needing guidance instructions, and those that I found on printables and github just did not cut it.
But, it seems, I was only seeing half the story.  There was no indication that you had put some further information on the wiki.
In fact there are pieces all over the place but nothing is linked together.  I know because I went looking for it.
The information about building the FW needs to be expanded.

The Videos are useless, please lean towards written text instructions.  I found only a couple of them and not the important ones regarding installing the firmware. And those instructions ought to be linked in the readme.md file and should be made big and obvious.

You need to do something about this OK !!!

Cheers,
Bryn Parrott

TheZeroBeast, 2022-07-14:
Again, if you’re not satisfied with my sole effort then feel free to start your own codebase from scratch. If you want to add more information then go right ahead via pull requests that I can review and merge as I see fit. I feel it’s explained sufficiently for people and videos are much better for the majority of people.

bryn51, 2022-07-16:
Hi Robert

I have been turning over in my mind what approach to take with this.

Your big effort to rewrite the printer and MMU code is noted.

But it’s just that which has me concerned. Looking in from the outside, I
see that Prusa have a code base put together by a team of programmers and
researchers dating back years and having thousands of hours hard testing,
initially in their own sizable print farm, and then years selling thousands
of copies worldwide into a hypercritical market.
On the other hand your effort does not (cannot) have that level of testing.
The goals and motivation for completely rewriting the code are not
transparent.

I look at the original problem I define by my own observation (lacking
knowledge of yours) as how to improve operation of the MMU having tendency
to jam on the filament path due to various causes but including
malformation of the filament tip. A solution being to have facility to cut
the filament tip upon unloading and this action can be performed in either
direction of selector travel.

This necessitates adjustment of MMU code and printer code to enable
automated cutting action.

My own approach to the need to adjust a proven code base to add some
functionality is to analyze it and make minimal changes if at all possible.
This task is aided if the existing code is written in an object oriented
manner. That is, deployment of classes with public and private methods.

Then it’s possible to deploy changes with reduced risk of destabilizing the
proven code base. And to test the code and have confidence in the results.

I have just recently been through such a process where my self assigned
task was to add sensors to an environment monitor system. The system is
complex, having multiple MCU’s networked using I2C.
Because the existing code was written in well formed object oriented c++
the task was relatively straightforward. I was able to add my pieces with
limited risk and deployable into other versions of the system without
changing their well tested operation.

So I have not yet looked at either (yours or Prusa) code base in any detail
to try to approach the problem from your perspective. I do not know your
coding background or skill level. It’s important not to be judgmental
because that can be destructive, not constructive.
I have only glanced at the source code to date.
I cannot believe that you actually rewrote large areas of the code, because
it’s a mammoth task. And large parts of that code base is devoted to
interpretation of gcode and physical operation of the printer including
stepper motor control that should not have had any need for change to
accommodate slightly different MMU operation. Although rewiring/refactoring
is expected.

But it’s as well not to prejudge these things. Better to research it before
opening my big trap further.

So, when I have the energy I’ll be doing just that. See you on the other
side.

Cheers
Bryn

My background fyi:
24 years working as a technical officer in a large well known telco. Cert
in Electronics and Communications.(included programming). Maker skills
acquired as part of basic training (4years). Ended as a Principal Technical
Officer.
Subsequently a further 20 years working as an analyst programmer on
contract and permanent staff in UK, NZ and Australia. C, C++, SQL and web
coding (HTML, cfml, JavaScript etc).
I’m retired from that now, but persuing electronics, IOT type projects
encompassing maker skills, electronics and coding as a hobby.

The 3D printing is added as a means to an end, not an end in itself; and
certainly I think of the printer as a tool to be used not a builder project
in itself. It will get chucked if it turns into one. I want to build
articulated robots.

bryn51, 2022-07-17:
Hi Robert,

OK I have quickly looked over the code for both Prusa MK3S and TZB and here are my findings:

  1.  Depth of analysis: shallow
  2.  Analyst Experience: Broad, Supervisory/Management, Architect, Analysis, Coding, Retired (age- related cognitive/memory impairment).
  3.  Analysis tool: Beyond Compare and VS Code/PlatformIO
  4.  Goal: Risk Assessment of applying TZB code to my printer

Findings:

  1.  Is the code formed of Well Formed Object oriented Code(?) No.  In both Cases the code might just as well have been written using K&R C code if judged on coding style alone. Whilst CPP code functions have been used there appears to have been some transition from older C only code to mixed CPP and C code, and the object orientation paradigm is completely absent.  This makes risk assessment more difficult.  This means that if a change is made to one part of the code base then it is possible to affect operation of other unexpected parts of the code base because the lack of object oriented classes and public/private methods does not protect against that happening.  The task of testing all possible program pathways is rendered almost impossible without an automated testing tool.
  2.  The TZB code base originates as a fork of Prusa 3.10.1 code, which is now several sub-versions behind the mainstream of Prusa code currently 3.11.1. Thus changes made to the Prusa code since release of 3.10.1 are omitted from the TZB Code as will future updates.  The Prusa changes are documented, but include both bug fixes and functionality changes.
  3.  The TZB Code changes are documented in-line by way of comments in the source code.  By this means its possible to discern that some proportion of the changes made the TZB relate to MMU and a further proportion related to  adding support for:

(a) BL Touch device
                (b) an enhanced display
(c) MK3 “bear”

Addition of code to support these devices is typically controlled by way of compiler switches found within  ‘configuration_prusa.h’ configuration file.  It looks like changes made this way include 100% of instances, but it is difficult to be certain without a “deep” analysis (i.e. more time spent).
Such deep analysis would necessarily include trial compilation of the source code with the compiler switches having different values and output hex files compared in binary, to assess how much of the final code changes when compared to the original prusa code compiled without any changes at all.

  1.  The TZB Code base includes a large proportion of unchanged code that is identical to the prusa code.  Thus it is inaccurate to describe the TZB changes as a complete rewrite.
  2.  Is the code backward compatible with unaltered Prusa Hardware ?  It is unclear if this is a necessary requirement.  The code would require extensive testing to determine if the statement is true or  false.
  3.  Is it essential to include the TZB Printer hardware in order to support TZB MMU2 firmware, or could the TZB MMU2 firmware be rebuilt to interact correctly with original Prusa Printer firmware ?  A: Unclear.  Testing indicates that todays version of TZB MMU2 firmware is incompatible with Prusa Mk3S firmware version 3.11.1

Summary and Conclusions
Having built the hardware for MMU2 dual cutter and TZB Extruder I thought I needed only to install the MMU2 TZB firmware.  But discovered its necessary to also install the TZB Printer firmware.  Upon investigation the latter also includes support for devices that are not in the frame so far as I am concerned.  But the additional changes add to risk.
The extensive changes make it likely unfeasible to merge changes made by Prusa back into the TZB code base.  If the changes made by TZB had been relatively minor or limited in scope to just a few files then it may have been feasible to simply merge the current Prusa Code base into the TZB for the bulk of files (with a few exceptions) , and thus gain benefit from progress made by Prusa Research in identifying and correcting bugs and adding enhancements.  (but that’s not the case).

Future action:
I do not know at this stage if it is possible to come up with another version of the MMU2 firmware code that remains compatible with Prusa Printer firmware, with minor changes to support only the enhanced MMU2 (without the additional devices).  But this should be investigated.
Do I have the time to do this ?  No.  Programming of printer firmware is not what I am about.  I just wish to use the thing as a means to an end, not the end in itself.

It’s possible to make multicolor prints on the Mk3S+ without the MMU2, by adding gcode that pauses the print and requests a filament change.  Or simply do without the ability to make multicolor prints.  Get another printer later that can do that : like the snapmaker (2 print heads) or the XL. (5 heads) or other Brand X printers.

The stepper motors found in the MMU2 could be applied to robotics projects so the hardware, so not completely useless.

Cheers and all the very best,
Bryn Parrott

From: Bryn Parrott ***@***.***>
Date: Saturday, July 16, 2022 at 10:10
To: TheZeroBeast/TZB-MMU2S-Firmware ***@***.***>
Subject: Re: [TheZeroBeast/TZB-MMU2S-Firmware] Readme.md file requires Update (Issue #233)
Hi Robert

I have been turning over in my mind what approach to take with this.

Your big effort to rewrite the printer and MMU code is noted and appreciated.

But it’s just that which has me concerned. <snipped>

But it’s as well not to prejudge these things. Better to research it before opening my big trap further.

So, when I have the energy I’ll be doing just that. See you on the other side.

Cheers
Bryn

TheZeroBeast, 2022-07-17:
Hi @bryn51,

Interesting analysis you have done here.

If you want the MMU2S to work as intended from PRUSA you need to take this up with them, I was in the initial shipment of units and as PRUSA did a poor job making a solid solution I came up with my own.

Primarily intended to have my farm stable with my hardware it usually only gets a lot of my attention when I see fit to update/change/troublesheet issues.

Like all other opensource projects you're very welcome to take PRUSA's codebase for printer/MMU2S and make the desired changes, good luck.

If you wish to contribute to this repo then like I've said you're welcome to, if I see benefit in your pull-requests then I will merge them.

bryn51, 2022-07-17:
Cheers Robert

I have little doubt that the TZB firmware meets your needs and you are able to maintain it for those purposes.

But it’s also clear that it’s not necessarily the case that you will keep it up to the same level as Prusa to meet needs of broader set of customers.

Not everyone has the capability to identify and correct bugs in a complex code base with which  do not not have  complete familiarity and is structured like it is today.

From my perspective I do not have an intention to take  on this task.

Instead I will retain the Prusa code and consider what options I have concerning the multicolor capability going forward. Whether to retain the MMU or remove it.

In a recent email exchange with Prusa support they indicated that PETG is not recommended for use with the MMU because of the string often found on the end of the filament upon unloading, and yet PETG is one of the filaments expressly supported. PETG is mainstream. TPU isn’t and I can forgive them for not supporting it, but not PETG.

And yet most of my filament stock is of PLA. It’s the only one I have in enough colors to consider a multicolor project.

To my mind I do not need to reach a decision right away (whether to retain the MMU or chuck it out), so I’ll just bide my time.

Cheers
Bryn

 Sent from my iPhone

gutschke, 2022-07-17:
This is such an odd discussion. It started out with a complaint that that the build instructions were a bit on the terse side. And I think I would agree. I have built my own image from the sources, and it took me a little bit to figure out how to set up an appropriate build environment. Fair enough.

But then it just veered off into personal attacks and questions about the legitimacy of the project. That's uncalled for. Robert addressed a very real need with this project. I still believe it is the most bang-for-the-buck to make the MMU2S more reliable. It's hard to keep up maintenance of a branch, if mainstream doesn't want to merge the changes. And I can see the other side of the story too. I am not surprised the Prusa prefers sticking to their own in-house development.

Everything considered, Robert has been extremely good about tracking Prusa's public releases. Credit where credit is due. And honestly, I would expect there to be very few big releases going forward. Prusa is putting a lot of work into the slicer, but the MK3S line is pretty mature at this point. They'll inevitably focus more of their work on future printer models.

So, enjoy the options that you have right now. You can use the stock firmware, which has improved over time. Or you can use Robert's firmware which has always been extremely reliable and user friendly, but might sometimes lag behind a little bit. I feel it is unrivaled at this point, and know which one I'll be using.

Now, if only it could make filament changes with my Revo 6 a little more reliable, then it would be perfect. But I am not sure that is a problem that can be solved in software. I took a stab at it, but the code flow is a little more complex than what I was able to sort out.

TheZeroBeast, 2022-07-17:
Thanks @gutschke, appreciate the support.

@bryn51, first off the only changes done in upstream branch after 3.10.1 which is one version to 3.11.0 (I don't pull RC) is as follows and not important feature changes that justifies the effort.
- New community translations
- Added LCD support for äöüß and μ
- SN check fixed
- Changes to on-screen messages
- Known issue (LCD glitches)

Now, you and your chat with PRUSA support, I think it natural to expect that a unit sold (and named) to be used with multiple materials at one time should do just that. They have told you **NO** this is not the case, it does PLA multicolour. **BS**, I have done plenty of TPU multicolour, TPU/PLA, TPU/ABS, TPU/PETG and PETG Multicolour prints with minimal tuning (after re-coding the MMU2S and it's comms protocol). They gave up on their code after they made tens of millions of pounds on the MMU2Ss, we're all pissed and only some make the most of it.