TheZeroBeast/TZB-MMU2S-Firmware #234: FW 4.0.2 Fails after upload. Not recognised by Einsy FW

bryn51, 2022-07-14:
**TZB-MMU2S Release Version**
4.0.2 BN380

**Describe the bug**
After FW installation (successful) using Prusaslicer: the firmware fails basic tests:  The main Einsy FW does not recognise the new firmware (it reports MMU version "N/A" and also I find that the sensor screen does not show the Finda probe, it shows just the PINDA and Fil Sensor (on the extruder).

**To Reproduce**
1. Edit Hex file to add the required line 
; device = mm-control 
as the first line of the file with one blank line following.

2. Connect the MMU to the PC using supplied USB cable (Usb-micro to USB-A)
3. Open Prusa Slicer, upload hex file using menu item under 'configuration'
4. Restart Printer by power cycle.
5. Wait for printer to initialize.  MMU makes strange noise.
6. On Einsy screen, navigate to support
7. scroll down to MMU : displays N/A instead of SW version
8. scroll down to "sensor info" Note display shows only 2 sensors instead of 3 expected.
9. go back to main menu.  Push Black Button, Navigate to "Load Filament". But find that this menu item is completely missing.
10. On Mac computer: Open 'System Information' app.
11. Navigate and select  USB.  Note that the USB 2.0 entry now shows 'Arduino Leonardo' instead of expected MMU port label.

7, 8, 9 & 11 are indicators that this MMU firmware appears not to be functioning.

**Expected behaviour**
See above

**Screenshots**
Not possible to add screenshots from the printer display.

**Serial Log**
(from Slicer)
Flashing Original Prusa MMU 2.0 Control, looking for VID/PID 0x2c99/3 or 0x2c99/4 ...
The Original Prusa MMU 2.0 Control device was not found.
The Original Prusa MMU 2.0 Control device could not have been found

**Additional context**
Its also made difficult to reinsall the old standard MMU firmware, because 'Arduino Leonardo' is not recognised by the slicer.
Whilst it is possible in the slicer to initially select the leonardo port, the slicer switches to automatic detection and then fails to detect the MMU signature, and the upload fails.
At this point I'm between a rock and. a hard place, forced to consider debugging this code that I've not analysed.
Or finding the original Prusa MMU FW code, building it and then uploading to the leonardo device.

TheZeroBeast, 2022-07-14:
Hey mate, my firmware is a full rewrite. You need the pair of both printer and mmu fw loaded for communication to work.

Also, [this](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/wiki/Flashing-Procedure) is the guide on loading back and forth.

closed 2022-07-14 by TheZeroBeast

bryn51, 2022-07-14:
Hi Robert,

I was unaware that the firmware update for MMU2 supporting dual cutting blade operation was for both printer and the mmu mcu.  I thought it was for the MMU only.

I also had not previously seen the videos depicting update of the firmwares back and forth.

For now, I have only installed the hardware on the MMU2, and not installed the updated extruder.  The latter are “ready to go” but I wanted to bed down and test the MMU part first.  Also I have never had the extruder apart before, since my printer came fully assembled.

That said, having watched the videos, I am partly aware of what needs to be done, but I am not ready and would like to go back to stock.
Please read the bottom paragraph regarding concerns about the printer firmware.

So, I have watched the video and read the instructions for going TZB > Stock.  I need only to revert the MMU to stock, since the printer FW was not changed.

When I go through the process (written instruction numbers are referenced): and my comments are below in Red:
I am using a Macbook Pro with Apple Silicon.  It does not have USB-A, so I am using a dongle.

  1.  Connect USB cable from Computer to MMU2S - Check
  2.  Open PRUSA Slicer (if not still open) - Check
  3.  Open Flash Firmware from Configuration menu - Check
  4.  Select the desired STOCK MMU2S firmware file – check (its version 1.0.6 the current one)
The dialog immediately changed to “autodetected”

  1.  Power ON printer - Check
  2.  Within 5s of powering ON printer click Flash  - I tried this and failed to hit the button quickly enough several times. 1 Second may be a better number to use.

  *   You should see the Computer rapidly progressing with flashing - OK
  *   Never disconnect or power OFF printer while Flashing - OK
  *   Computer should report success - OK
Eventually, after trying several times, it worked.

  1.  Power OFF Printer (PSU toggle switch is fine again)
  2.  Disconnect USB from MMU2S
  3.  Wait 10s
  4.  Power ON printer
OK, now to revisit this Printer firmware update.
I have the concern that using your custom printer firmware then precludes my gaining advantage from any update made by Prusa to the stock firmware.  Recent updates have been inconsequential, but you never know if they decide to make a real improvement or a bug fix.
How do you respond to that concern ?
Cheers, and thanks for responding quickly to my previous concerns.
Bryn Parrott

TheZeroBeast, 2022-07-14:
If you can’t follow you can always make pull requests with suggestions that I can review.