TheZeroBeast/TZB-MMU2S-Firmware #237: There is the "normal" MMU2S firmware version missing for 4.0.2 only dual blades are availale. Please :)

Ic33s, 2022-08-22:
Ensure you're running the latest **RELEASE** and you've checked the **Wiki** for answers before reporting. Any issues reported regarding previous **RELEASES** will automatically be closed and not responded to, this ensures the developer maintains focus & streamlines the development process.

**TZB-MMU2S Release Version**
4.0.2

**Describe the bug**
Not hex release of singleblade (stock selector) available.

Would be great. I've nealy given up on MMU2S. Your firmware is my last hope, with the stock firmware my MMU2S the filament loading is completely scewed up (I've multiple times checked all sensors). :)

Tony763, 2022-08-22:
Hi @Ic33s, use `TZBDualblade`. It is compatible with stock (single).
 
[image]

Ic33s, 2022-08-22:
Thank you for your fast reply. Just flashed it, unfortunatly no success either. I don't get any filament loaded (since hours) and I don't understand whats the cause of the problem. But that's not your problem :) ... The MMU2S made problems from the start and seems a waste of money unfortunately. Best

closed 2022-08-22 by Ic33s

Tony763, 2022-08-22:
@Ic33s did you also flash a Printer? I agree that MMU2S is pain in a**, but with this incredible FW and few hours of try/error it can run pretty good. Does printer write something on display or status LEDs on MMU?

TheZeroBeast, 2022-08-22:
Thanks @Tony763 for helping mate.

@Ic33s, stick with it, the MMU2S can be setup stable with minimal interventions when in use. Double check all the hardware and don't over torque the tension bolts on the MMU2S, people usually do.

Ic33s, 2022-08-23:
@Tony763 @TheZeroBeast: Thank you for your kind words. After nearly 2 years of fighting with the MMU2S I was finally in the mood to take it of and to bury it. But I got it running again yesterday right after midnight.

@TheZeroBeast I know that can make more trouble than help at the MMU2S itself but also at the extruder side. 
The Phaetus extruder I've tried the last few month decided to have some heatcreap and caused a problem. Yeah, the first time in 2 years not the MMU2S makes the trouble itself. @TheZeroBeast your firmware was reporting a clog - the stock firmware simply has not so much feedback. It's more like: "Hey I'm not working ... Here, I have a riddle for you". So thank you very much for clear status messages.
I think I stick with your firmware and I will finally think about changing the whole printhead by using your design or the vertex. At the moment my printhead is still mostly stock beside the fan duct and some little changes made by myself (with a longer PTFE path and M10 fast coupler - which already helped a lot). Btw. how do you plan to proceed with your firmware? Will there be updates, when Prusa is updating the base you forked off?

Best