TheZeroBeast/TZB-MMU2S-Firmware #238: No precompile MMU2 file for 4.0.2

floridaservices, 2022-08-23:
**Is your feature request related to a problem? Please describe.**
Precompiled file for flashing MMU2S to version 4.0.2 is not in list of available files to download

**Describe the solution you'd like**
Suggest making available precompiled MMU2 firmware 4.0.2 hex

**Describe alternatives you've considered**
I currently run 4.0.1 with no issues but others have asked me about 4.0.2 availability so I am reporting this on their behalf, since you need the file for MK3S and MMU2.

Tony763, 2022-08-23:
Hi, all files are listed in release.

[image]

Stock MK3S Printer [TZB-MK3S-4.0.2-BN2528.hex](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/releases/download/4.0.2/TZB-MK3S-4.0.2-BN2528.hex)

Stock MMU2: [TZB-MMU2S-TZBDualBlade-4.0.2-BN380.hex](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/releases/download/4.0.2/TZB-MMU2S-TZBDualBlade-4.0.2-BN380.hex)

floridaservices, 2022-08-23:
It would seem from the file name that TZB-MMU2S-TZBDualBlade-4.0.2BN380.hex is not for the stock configuration.   If you look at 4.0.1 file TZB-MMU2S-4.0.1-BN376.hex is listed in place of TZB-MMU2S-TZBDualBlade-4.0.2BN380.hex so perhaps you are right, but it's not obvious to first time users.  I think it would help people new to the TZB ecosystem if the file was named accurately.

Tony763, 2022-08-23:
That is why this is in release description:

[image]

Stock/Single blade and TZB Dual blade are compatible and thus use same file

For Dual Cutter there is separate file.

Agree, that it can be little confusing for newbie, but if unsure it can be easily found in closed issues.

jaddel, 2022-12-15:
I just ran into this and was very happy for this created issue. :)